pulp-platform/pulpissimo #5: syntax error with CDN incisive toolchain

lighteningfingers, 2018-02-21:
mac_streamer.sv line 76:
    .tcdm               ( tcdm.master[0:0]       ), // this syntax is necessary as hwpe_stream_source expects an array of interfaces

modport definition for tcdm port is already defined in line 43, solution:

 .tcdm               ( tcdm[0:0]       ), // this syntax is necessary as hwpe_stream_source expects an array of interfaces

hwpe_ctrl_ucode.sv line 127
  string str = "";

error - multiple drivers to always_comb output var str
solution:
  string str;

FrancescoConti, 2018-02-21:
Thanks for pointing out issues with Cadence IUS - both errors are invisible to ModelSim.
The first error I spotted a couple of days ago, it is fixed already if you update the IPs; the second one I missed... I'll fix it ASAP.

PS: did you generate Cadence scripts with the `ipstools` or manually?

closed 2018-02-21 by FrancescoConti

lighteningfingers, 2018-02-21:
the scripts were generated manually.

I'd like to enquire about the use of DPI from within the top level harness...
1 - why choose DPI and not just write the whole harness in SV (for example like a UVM env)
2 - should the DPI calls be tool compatible across vendors?

FrancescoConti, 2018-02-21:
@haugoug is the right person to answer DPI-related questions; however I can already say that our "official" support is only for QuestaSim at the moment. That said, some of our partners are/will be using Cadence - so it is likely that the code is designed to work in both setups.

Why not SV/UVM? Part of the choice of DPI is related to the fact that we want to reuse a lot of our setup for our virtual platform (not yet released, should be this year). That said, we do have some (small) SV verification in place and we're likely going to strengthen it in the future - [Ariane](https://github.com/pulp-platform/ariane) is using UVM, while I feel for PULP/PULPissimo we're steering towards simpler SV-based verification.
Personally coming from a HW design perspective, I feel UVM tries to solve a problem at too general a level, thus the necessity of a lot of boilerplate code and hugely complex class hierarchies. It quickly becomes unmanageable in a small academic setting like ours (ETHZ + UNIBO).

lighteningfingers, 2018-02-21:
thanks for the contact WRT the DPI, I may make contact in the next few days.

My reference to UVM was purely as an example of a SV only environment, and was in no way an endorsement.  (As you point out, UVM can become unmanageable unless coding guidelines are adopted to keep a uniform flow).

I understand the inheritance of a lot of the harness code from the virtual platform env, however I still feel that this is a significant barrier to wider adoption using more common tools (certainly CDN IUS and SYN VCS are widely used in industry).
I would even suggest a verilator setup would be more approachable rather than a specific non-free commercial tool chain..... I don't have access to any Mentor tools, and this is a barrier to our adoption.
I hope to have some good (CDN IUS) news soon, at the moment I have a design that compiles and elaborates.