pulp-platform/pulpissimo #11: RISCY+FPU Dhrystone

lfernand4, 2018-05-03:
First of all, great work! 

I run dhrystone on RISCY and RISCY+FPU . Normaly I should get the same result because DHRYSTONE does not use floating point. 
But I get 1.10 for RISCY and 1.06 for RISCY+FPU. 
With Coremark I have the exact same result, 3.13 for RISCY and RISCY+FPU.
Do you know why ?

Thanks in advance

haugoug, 2018-05-03:
Indeed it is normal that coremark is having the same result as there is no float but I have no idea why dhrystone is different.
If you want to investigate you can generate the trace files for both cores (the files called trace_core_00_0.log) and generated inside the build folder. These files gives the list of instructions which were executed plus a lot of additional information like timestamp and cycles. They are really easy to parse from python and is a good way to compare 2 executions. Also if you look at the second column (Cycles), you can see when the core is stalled.

lfernand4, 2018-05-04:
I see some floating point instruction and INVALID instruction.  Do you know why ?
All the instructions are in this online-excel file : https://1drv.ms/x/s!AipdcGmJxDv3rS0gNiln-mmVS39v.

haugoug, 2018-05-04:
There seems to be some floats in the C code. You can take the PC where you see an FPU instruction and then look into the disassembled binary which C code corresponds to it. You can use this command to disassemble:
make dis disopt="-d -l" | less

lfernand4, 2018-05-14:
Do you understand why "fcvt" is an "INVALID" instruction ( ri5cy+FPU)

[image]

[image]

ethjenkins, 2018-05-16:
It just seems that the tracer module is not recognizing this instruction but as it does not generate any illegal instruction, it means the execution was fine. We will fix the tracer soon, let us know if you see others.

closed 2018-06-07 by FrancescoConti

sharayup12, 2020-10-07:
@lfernand4 i want to run the dhrystone on RV32m1 board which dhrystone source code did you used? and what changes do i have to make in odrder to run it on the riscy core?