pulp-platform/pulpissimo #12: Cannot find vopt

lstrz, 2018-05-05:
I've followed the instructions to run the simulation: clone, update-ips, source vsim.sh, run make clean lib build opt. At some point during make I get an error `Error: cannot find "/opt/intelFPGA/17.1/modelsim_ase/bin/../linux/vopt"`

I have installed the latest version of Altera's/Intel's ModelSim that I could find online. Version 17.1 of Quartus, but the ModelSim version is 10.5b. `vopt` executable does exist, but is a symlink to the `vco` script which, after being ran as `vopt`, resolves to the path in the error message, where `vopt` doesn't exist. No other `vopt` executable exist anywhere.

Am I doing something wrong? Can `vopt` be replaced with some other executable? README.md says that pulpissimo was tested with ModelSim 10.6b, but I can not find that version anywhere online. Is it Altera's/Intel's ModelSim or Mentor's or another one?

FrancescoConti, 2018-05-05:
Hi @lstrz, ModelSim/QuestaSim is Mentor's version (i.e. the original one). As far as I know Altera's one is licensed from Mentor (https://www.altera.com/products/design-software/model---simulation/modelsim-altera-software.html) -- unfortunately I have no idea why this is missing `vopt`. I guess ModelSim-Altera is meant to be used in a fairly constrained setup and does not support well our flow.

The explicit optimization step can be bypassed, however, so what you can try to do is launch directly `vsim` (jumping the `make opt` stage), but you will have to edit the `sim/tcl_files/run.tcl` file and the ones called by it to add all the libraries explicitly like is done in the `sim/tcl_files/rtl_vopt.tcl`. Let us know if you manage to solve the issue or you need further help.

lstrz, 2018-05-14:
I've been looking at Mentor's ModelSim for a while now, but have failed to find a Linux download. Nowhere on their webpage do they say anything about Linux, and all the downloads I could find are .exes for Windows. Could you, please, verify the version that you have available, if you're running it on Linux?

FrancescoConti, 2018-05-14:
To be honest, I have never run Mentor ModelSim on Windows and was unaware that it existed. ModelSim/QuestaSim is a standard (and rather expensive) EDA package and I have run it on RHEL / CentOS 6 and 7, as well as on Ubuntu 16.04. In fact we do not support Windows at all for PULPino/PULPissimo/OPENPULP.

If you are unavailable to get access to Mentor ModelSim via a corporate or academic network, it's probably best to try and adapt the scripts following the second part of my previous comment. Let me know if I can provide further help with that.

boehmerst, 2018-05-15:
Hi @lstrz I managed to get pulpissimo running with Altera/Intel Modelsim 10.6c Starter on Ubuntu 16.04. There are some workarounds required as Ubuntu is not supported officially. Intructions can be found here: https://lvoudouris.com/installing-xilinx-vivado-2016-4-and-intel-modelsim-starter-edition-16-1-on-64-bit-ubuntu-16-10/

To get it working I had to hack the flow as @FrancescoConti explained. Additionally to the libraries to be added, the top-level testbench has to be changed from `vopt_tb` to `pulp_tb` in `run.tcl`. Finally I had to hack the `plp_rtl_runner.py` in the SDK because Altera Starter is available as 32 Bit version only. After that I successfully run the `Hello World` example for both the pulpissimo and the pulp platform.

Modelsim Altera Starter is limited in features e.g. no `vopt, 10.000 lines of code, 3.000 instances, 32 Bit only`. That is why simulation performance is extremely low. Actually, it is not useable for designs as complex as pulpissimo. 

Unfortunately the `accelerator/hwme` example fails! I nailed down the root cause to an incorrect bit with of the hwpe_stream_fifo. For whatever reason the parameter assignment of 68 Bit does not propagate down the hierachy and the actual instance `i_fifo` in file `hwpe_stream_tcdm_fifo_store.sv` is falling back to the parameter default which is 32 Bit. This seems to be a Modelsim issue as the code looks straight forward to me. Forcing this parameter with the deprecated `defparam` syntax fixes the issue. Also working around the `generate` statement up in the hierarchy in `hwpe_stream_sink.sv` fixes the issue. Both are dirty hacks I would not suggest to introduce. It might be the case that there are similar issues with other IPs also. So I do not trust the Modelsim Altera Starter / Ubuntu 16.04 flow. I would be interessted if anyone has experienced similar with the Altera/Intel Modelsim FPGA Edition or Modelsim DE or even QuestaSim respectively.

My goal is to get pulpissimo working with the verilator compiler which is painful as verilator is very pedantic. A lot of the IPs cause compilation errors. Do you have any plans to support verilator in future? I think there are a lot of open source enthusiasts (including me) which do not have access to expensive tooling.

FrancescoConti, 2018-05-15:
@boehmerst regarding the `hwme` test -- that's interesting, because the same thing happened to me with the same FIFO at a certain point (on an internal version of the platform). You actually made me remember the fix was never backported here, even if it is present in the multi-core `pulp`. The fix is not related to the test or the `hwpe-stream`, but to the optimization of parameters. It is there also in QuestaSim.

For reference, here is the explanation (I will commit the fix now to `pulpissimo`):
```
This commit fixes the incorrect optimization of some modules.
HWPEs appearsnot to work because several parameters in
submodules are not bound correctly (the default value is used instead).
This causes subtle problems in simulation. Moreover, it appears the
vopted version is not actually being used.

The fix follows what is done in OPENPULP by using vopt on the entire
design (including the testbench), but leaving the parameters in the
top-level tb floating.
```
Thanks 🥇👍 

Regarding your second question: I am not sure that we ever have the "bandwidth" to do a proper porting to Verilator, and keep it working, but if you want I think you can open a Verilator-related issue in PULPissimo and post a few of the errors you are getting. This should at least significantly reduce the pain, and you can also contribute fixes back so that we can start integrating them.

FrancescoConti, 2018-05-15:
@boehmerst I have to correct myself: the fix is already in PULPissimo, but by using the "hack" above to get PULPissimo to work in Altera ModelSim, the parameters are not propagated correctly (essentially the same that happened before `-floatparameters` was added in the other case I mentioned).

So I think for now the only thing to do is keep using the two dirty hacks you mentioned :(

boehmerst, 2018-05-16:
@FrancescoConti, Thanks for your feedback! Curious issue, in fact all modules (from the `stream` IP) I instantiated (just for debugging) the on the same hierarchical level of the suspect FIFO showed the same incorrect behavior whereas a simple `clock_divider` from `common_cells` worked as expected. 

I will keep on debugging the Verilator errors. For now most are straight forward (usually issues with blocking vs non-blocking assignments to the same variable when using structs). But there are a lot of those. This is a known Verilator limitation which is unlikely to be fixed soon. They are easy to work around, however there is a certain risk to break things and there are not many module level test benches. The `hwme` is the first IP I got "verilator clean" and I will port the `accelerator` example to a simple module test bench next. In case I get things up and running I will be happy to contribute back!

lstrz, 2018-05-18:
My university does have a license for Questa/ModelSim. Indeed, the non-free version includes vopt. That has solved all my issues. Before @boehmerst's input, I did not know for sure whether the free version is supposed to include vopt. Thanks for input everyone!

Might not hurt to mention this in the readme.

FrancescoConti, 2018-05-30:
@lstrz I just made the `README.md` more precise on this point. Thansk for the input from all of you!

Vaktor777, 2018-07-12:
Hi, @boehmerst 
could you show me your run.tcl file? I have some troubles to work around vopt flow optimization. How do you include libraries?
Thanks.

FrancescoConti, 2018-08-03:
Hi @Vaktor777 , can you push the changes that you made to a branch/fork? I think it would be nice to merge them in.

Vaktor777, 2018-08-03:
Hi @FrancescoConti , yes i can push commit with run.tcl, but plp_rtl_runner.py is from pulp-sdk, give me the name of the branch.

FrancescoConti, 2018-08-03:
Just fork both projects and submit a Pull Request named "ModelSim for Altera" (assigned to me).

closed 2018-08-03 by FrancescoConti

Vaktor777, 2018-08-03:
@FrancescoConti 
i cant assign to you, but i did pull requests to pulpissimo and runner repository named "ModelSim for Altera"

Alessandro0110, 2018-10-24:
Hello @lstrz your university has the Mentor version of Modelsim or the Altera version of Modelsim?

lstrz, 2018-11-13:
The Mentor version.