pulp-platform/pulpissimo #13: coremark micro-riscy

lfernand4, 2018-05-15:
I run coremark on riscy, zero riscy and riscy+FPU with no error, but with microriscy I have the following error : 
[image]

Do you know why ?

FrancescoConti, 2018-05-15:
There seems to be a problem in linking. Maybe @haugoug can detect which one :)

ethjenkins, 2018-05-16:
I had a quick look but I haven't figured out yet what the problems is. I will investigate more and let you know.

lfernand4, 2018-07-23:
I still have the same error. It run correctly with -02 -g -falign-functions=16 -funroll-all-loops  but it break with -03 -g -falign-functions=16 -funroll-all-loops. If I use only -03 it work.
[image]

closed 2019-04-25 by lfernand4