pulp-platform/pulpissimo #14: error while trying to run the hello example

sharinoz, 2018-05-22:
hi all
I am having some trouble while I try to make the hello example work
I did everything by the order, and when I go to the hello directory and enter:
make clean all run
this is what I get:
plpflags gen  --output-dir=/home/rotemshahar/pulpissimo/sim/pulp-rt-examples/hello/build/.home.rotemshahar.pulp-sdk --makefile=/home/rotemshahar/pulpissimo/sim/pulp-rt-examples/hello/build/.home.rotemshahar.pulp-sdk/config.mk    --app=test --out-config=/home/rotemshahar/pulpissimo/sim/pulp-rt-examples/hello/build/.home.rotemshahar.pulp-sdk/config.json
make: plpflags: Command not found
make: *** No rule to make target 'clean'.  Stop.

please help me to figure out what has gone wrond
thanks

FrancescoConti, 2018-05-22:
Have you installed the SDK? Is it correctly set in your environment, as per the `README.md`?

FrancescoConti, 2018-05-30:
I will close the issue as there has been no update on the initial problem.

closed 2018-05-30 by FrancescoConti

dogukan-bicer, 2024-05-05:
aynı hatayı bende aldım --> dogukan@dogukan-VirtualBox:~/pulpissimo$ cd pulp-runtime-examples/hello

make clean all run

plpflags gen --input=/home/dogukan/pulp-configs/configs/chips/pulpissimo/pulpissimo.json  --config=rt/start-all=true  --config=**/rt/fc-start=true --output-dir=/home/dogukan/pulpissimo/pulp-runtime-examples/hello/build/.home.dogukan.pulp-configs.configs.chips.pulpissimo.pulpissimo.json --makefile=/home/dogukan/pulpissimo/pulp-runtime-examples/hello/build/.home.dogukan.pulp-configs.configs.chips.pulpissimo.pulpissimo.json/config.mk    --app=test 

/bin/sh: 1: plpflags: not found

make: *** No rule to make target 'clean'.  Stop.  
 
nasıl düzeltebiliriz?

dogukan-bicer, 2024-05-05:
I got the same error too --> dogukan@dogukan-VirtualBox:~/pulpissimo$ cd pulp-runtime-examples/hello

make clean all run

plpflags gen --input=/home/dogukan/pulp-configs/configs/chips/pulpissimo/pulpissimo.json --config=rt/start-all=true --config=**/rt/fc-start=true - -output-dir=/home/dogukan/pulpissimo/pulp-runtime-examples/hello/build/.home.dogukan.pulp-configs.configs.chips.pulpissimo.pulpissimo.json --makefile=/home/dogukan/pulpissimo /pulp-runtime-examples/hello/build/.home.dogukan.pulp-configs.configs.chips.pulpissimo.pulpissimo.json/config.mk --app=test

/bin/sh: 1: plpflags: not found

make: *** No rule to make target 'clean'. Stop.
 
how can we fix it?