pulp-platform/pulpissimo #16: divergence float FPU

lfernand4, 2018-05-28:
I run this short C code with RISCY, RISCY+FPU and GCC : 
[image]
Result with GCC and RISCY (No FPU) : 
[image]
Result with RISCY+FPU : 
[image]

Can you explain to me this error with the FPU ?

FrancescoConti, 2018-05-30:
I think maybe @stmach can help you

stmach, 2018-05-30:
Hi! Thank you for raising an issue.
What platform (rtl simulation, virtual platform) do you use to run the program?
There seems to be a mismatch between the program or model depending on whether FP emulation in software is done or not.
What strikes me as interesting is that the output of the second code contains values like `46.99999936` which is not a value that can be represented in single-precision floating-point.
I see following factors that could play a role in this issue:
- You use a custom print function for floats where potentially some implicit cast to `double` could happen
- The initialization of your vectors is done by implicit casts from `int`
- If you enable hardware-floats, the compiler might do be trying to optimize the code and add some static initialization using `double`

In order to try to figure out the source of the issue, please try the following and report any interesting findings:
- Change initialization to explicitly assign `float` values instead of integers in the `generateur` function
- Try using `printf` with the `%f` switch to print float values (if supported by the runtime). However printf's %f switch operates on doubles so we might see wrong values here too.
- Print out the bit-patterns of your vector to see what's actually stored in there, by getting the patterns trough the hacky '*(int *) &x[n]` cast on the vector elements

lfernand4, 2018-05-30:
I run program with the RTL simulation platform.
Explicitly assign float values : Same Issue
printf with the %f : 
[image]
Print out the bit-patterns : 
[image]

stmach, 2018-05-30:
Looking at the bit patterns (easier seen when looked at in hexadecimal), the correct representation of the floats are actually stored in the variable.
Thus the error likely lies in the `printfloat` function. Be aware that you are trying to print out single-precision floating-point values with 10 digits of accuracy while single-precision floats only give you around 7 usable decimal digits of accuracy. I don't know where exactly the error gets in but it might be the compiler that statically inserts some double precision values because you seem to need more precision than float offers.
Try debugging your print function by printing the intermediate values it computes when printing the float value 47.0 for example and see at which point you don't receive the expected output.

lfernand4, 2018-05-30:
And why can't I use printf with %f ?

stmach, 2018-05-30:
Likely the version of printf in the runtime does not include the %f flag.
Pinging @haugoug, we will know.

lfernand4, 2018-05-30:
Thank you for all those informations

closed 2019-04-25 by lfernand4