pulp-platform/pulpissimo #20: Consider using commit hashes and tags instead of branch labels for stable version.

bluewww, 2018-06-16:
As is mentioned in the README, the master branch of IP's is not necessarily stable. So I think that ips_list.yml of this top level project should not reference the master branch of pulp_soc but a tag/commit. References to branches should be only used in development, which could be mentioned in README. Forcing a change in the ips_list.yml if someone wants to update their IP to a new version makes it apparent which change introduced a regression.
Unfortunately It looks like only commit hashes are truly immutable. I observed changes before that just moved the tags/pulpissimo-v1.0 to another commit which I would consider bad practice.

FrancescoConti, 2018-06-16:
Mmm, I get your point, I am not sure I agree 100% though as the pulp_soc is not meant to be used for development, at all. I will think about it for the future.

bluewww, 2018-06-16:
Sometimes changes to `pulp_soc` are needed for example to support a new peripheral. I guess in that case one needs to fork `pulp_soc` anyway and create a new branch which is then tracked in this repository vis its `ips_list.yml`. But even in that case the fork still evolves with changing ips, since sometimes people force push the tag `tags/pulpissimo-v1.0` onto a new commit, which I think is unexpected behaviour. Maybe atleast this practice should be discouraged in the `README`.

closed 2019-03-12 by bluewww