pulp-platform/pulpissimo #23: error while Building the RTL simulation platform

Vaktor777, 2018-06-27:
Hello 
following the instructions https://github.com/pulp-platform/pulpissimo/blob/master/README.md I run
.../pulpissimo/sim$ LANG=C make build
  Building tb/tb
** Warning: ../rtl/tb/dpi_models/dpi_models.sv(176): (vlog-2240) Treating stand-alone use of function 'dpi_jtag_bind' as an implicit VOID cast.
** Warning: ../rtl/tb/dpi_models/dpi_models.sv(188): (vlog-2240) Treating stand-alone use of function 'dpi_ctrl_bind' as an implicit VOID cast.
** Warning: ../rtl/tb/tb_driver/tb_driver.sv(52): (vlog-2250) Function "register_qspim_itf" has no return value assignment.
** Warning: ../rtl/tb/tb_driver/tb_driver.sv(59): (vlog-2250) Function "register_jtag_itf" has no return value assignment.
** Warning: ../rtl/tb/tb_driver/tb_driver.sv(64): (vlog-2250) Function "register_ctrl_itf" has no return value assignment.
** Warning: ../rtl/tb/tb_pulp.sv(216): (vlog-2244) Variable 'i_tb_driver' is implicitly static. You must either explicitly declare it as static or automatic
or remove the initialization in the declaration of variable.
** Warning: ../rtl/tb/tb_pulp.sv(218): (vlog-2240) Treating stand-alone use of function 'register_qspim_itf' as an implicit VOID cast.
**** Error: ../rtl/tb/tb_pulp.sv(218): (vlog-2110) Illegal reference to interface "qspi_0_csn_1".
** Error: ../rtl/tb/tb_pulp.sv(218): (vlog-2110) Illegal reference to interface "qspi_0_csn_0".**
** Warning: ../rtl/tb/tb_pulp.sv(219): (vlog-2240) Treating stand-alone use of function 'register_jtag_itf' as an implicit VOID cast.
** Warning: ../rtl/tb/tb_pulp.sv(220): (vlog-2240) Treating stand-alone use of function 'register_ctrl_itf' as an implicit VOID cast.
make[2]: *** [/home/barbarossa/workspace/pulpissimo/pulpissimo/sim/modelsim_libs/tb_lib/tb.vmake] Error 2
make[1]: *** [build] Error 2
make: *** [build] Error 2
please help me.
thanks

haugoug, 2018-06-27:
This is due to the last commit, I will have a look.
For now checkout this version: 
git e7ddd08b75dea619df92a5d80bf304e414c1bb94

Vaktor777, 2018-06-27:
yes it works, thanks

haugoug, 2018-06-27:
I didn't manage to reproduce it but I've fixed a few warnings, could you retry with latest master and tell me if it is better ?

FrancescoConti, 2018-06-28:
The issue is definitely not closed.

Vaktor777, 2018-06-28:
@haugoug , this is latest master
make build
  Building L2_tcdm_hybrid_interco/soc_interconnect
  Building adv_dbg_if/adv_dbg_if
  Building apb2per/apb2per
  Building apb_adv_timer/apb_adv_timer
  Building apb_fll_if/apb_fll_if
  Building apb_gpio/apb_gpio
  Building apb_node/apb_node
  Building apb_interrupt_cntrl/apb_interrupt_cntrl
  Building axi_node/axi_node
  Building axi_slice/axi_slice
  Building axi_slice_dc/axi_slice_dc
  Building timer_unit/timer_unit
  Building common_cells/common_cells_all
  Building fpu/fpu
  Building jtag_pulp/jtag_pulp
  Building riscv/riscv
** Warning: ../ips/riscv/./rtl/riscv_controller.sv(837): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/riscv/./rtl/riscv_controller.sv(839): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/riscv/./rtl/riscv_controller.sv(31): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
  Building riscv/riscv_vip_rtl
  Building riscv/riscv_regfile_rtl
  Building zero_riscy/zeroriscy
** Warning: ../ips/zero-riscy/zeroriscy_controller.sv(516): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/zero-riscy/zeroriscy_controller.sv(33): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/zero-riscy/zeroriscy_debug_unit.sv(481): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
  Building zero_riscy/zeroriscy_vip_rtl
  Building zero_riscy/zeroriscy_regfile_rtl
  Building scm/scm
  Building generic_FLL/fll
  Building tech_cells_generic/tech_cells_rtl
  Building udma_core/udma_core
  Building udma_uart/udma_uart
  Building udma_i2c/udma_i2c
  Building udma_i2s/udma_i2s
  Building udma_qspi/udma_qspi
  Building udma_sdio/udma_sdio
  Building udma_camera/udma_camera
  Building hwpe_ctrl/hwpe-ctrl
  Building hwpe_ctrl/tb_hwpe_ctrl
** Warning: ../ips/hwpe-ctrl/tb/tb_hwpe_ctrl_seq_mult.sv(78): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/hwpe-ctrl/tb/tb_hwpe_ctrl_seq_mult.sv(19): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
  Building hwpe_stream/hwpe-stream
** Warning: ../ips/hwpe-stream/rtl/hwpe_stream_interfaces.sv(46): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/hwpe-stream/rtl/hwpe_stream_interfaces.sv(46): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/hwpe-stream/rtl/hwpe_stream_interfaces.sv(51): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
  Building hwpe_stream/tb_hwpe_stream
  Building hwpe_mac_engine/hw-mac-engine
** Warning: ../ips/hwpe-mac-engine/rtl/mac_engine.sv(234): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/hwpe-mac-engine/rtl/mac_engine.sv(234): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/hwpe-mac-engine/rtl/mac_engine.sv(238): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
  Building pulp_soc/pulp_soc
  Building pulp_soc/udma_subsystem
  Building pulp_soc/fc
  Building pulp_soc/components
  Building pulp_soc/components_rtl
  Building pulp_soc/components_behav
  Building tbtools/tbtools
  Building tb/tb
** Error: ../rtl/tb/tb_pulp.sv(218): (vlog-2110) Illegal reference to interface "qspi_0_csn_1".
** Error: ../rtl/tb/tb_pulp.sv(218): (vlog-2110) Illegal reference to interface "qspi_0_csn_0".
make[2]: *** [/home/barbarossa/workspace/pulpissimo/pulpissimo/sim/modelsim_libs/tb_lib/tb.vmake] Error 2
make[1]: *** [build] Error 2
make: *** [build] Error 2

FrancescoConti, 2018-06-28:
@Vaktor777 @haugoug I temporarily reverted the change that caused this issue. The `master` should work now...

closed 2021-02-02 by meggiman