pulp-platform/pulpissimo #25: verilator compiler

Vaktor777, 2018-07-04:
Hello, 
I'm trying to run  .sv files in verilator. I have some difficulties with linking . First i add 'include' include "pulp_interfaces.sv" in ../pulp_soc/rtl/pulp_soc/pulp_soc.sv. And now 
%Warning-REDEFMACRO: ../includes/periph_bus_defines.sv:19: Redefining existing define: NB_MASTER, with different value: 11
%Warning-REDEFMACRO: Use "/* verilator lint_off REDEFMACRO */" and lint_on around source to disable this message.
%Warning-REDEFMACRO: ../includes/soc_bus_defines.sv:116: Previous definition is here, with value: 4
%Warning-STMTDLY: ../../ips/pulp_soc/rtl/components//freq_meter.sv:36: Unsupported: Ignoring delay on this delayed statement.
%Error: ../../ips/pulp_soc/rtl/pulp_soc//pulp_soc.sv:410: Pin not found: test_cgbypass_i
%Warning-IMPLICIT: pulpissimo.sv:782: Signal definition not found, creating implicitly: s_boot_l2
%Warning-IMPLICIT: pulpissimo.sv:1029: Signal definition not found, creating implicitly: s_so
%Error: Exiting due to 1 error(s), 5 warning(s)
%Error: Command Failed /usr/local/bin/verilator_bin --cc pulpissimo.sv --top-module pulpissimo -y ../includes -y ../../ips/jtag_pulp/src/ -y ../../ips/pulp_soc/rtl/pulp_soc/ -y ../../ips/pulp_soc/rtl/components/
Maybe I 'm going the wrong way.  Could you help me?
Thanks.

FrancescoConti, 2018-07-04:
Hi @Vaktor777 , the issue is probably due to a mismatch in the definition of the `test_cg_bypass_i` port in `pulp_soc.sv` line 410. It is a bit weird because this specific problem should be also visible in ModelSim, what commit of `pulpissimo` and `pulp_soc` are you pointing to? Check with `git log`.

Vaktor777, 2018-07-05:
Hi, @FrancescoConti 
`/pulpissimo/rtl/pulpissimo$ git log -1
commit 0bf08d0f2797c9134f1ca0fb046f9a0a625a7e98
Merge: d204c4a ae58559
Author: Francesco Conti <FrancescoConti@users.noreply.github.com>
Date:   Thu Jun 28 14:50:06 2018 +0200

    Merge pull request #24 from pulp-platform/fix-issue-23
    
    Merging after @haugoug 's further commit.

/pulpissimo/ips/pulp_soc/rtl/pulp_soc$ git log -1
commit 4179bbfd1edf0cb53b1f07a6eb6c21f8a21b96ac
Author: Pasquale Davide Schiavone <pschiavo@iis.ee.ethz.ch>
Date:   Thu Jun 21 11:57:52 2018 +0200

    Update ips_list.yml
`

Vaktor777, 2018-07-31:
hello, i'm still trying to run pulpissimo in verilator, now i have a problem
`/pulpissimo$ verilator -DPULP_FPGA_EMUL --error-limit 100 -cc ./rtl/pulpissimo/pulpissimo.sv -I./rtl/pulpissimo/ -I./rtl/includes -I./ips/jtag_pulp/src/ -I./ips/pulp_soc/rtl/pulp_soc/ -I./ips/pulp_soc/rtl/ -I./ips/pulp_soc/rtl/components/ -I./ips/axi/axi_slice_dc/ -I./ips/common_cells/ -I./ips/L2_tcdm_hybrid_interco/TB/ -I./ips/L2_tcdm_hybrid_interco/RTL/XBAR_L2/ -I./ips/L2_tcdm_hybrid_interco/RTL/XBAR_BRIDGE/ -I./ips/L2_tcdm_hybrid_interco/RTL/ -I./ips/L2_tcdm_hybrid_interco/RTL/axi_2_lint/ -I./ips/axi/axi_slice/ -I./ips/timer_unit/ -I./ips/apb/apb_node/ -I./ips/adv_dbg_if/rtl/
%Warning-PINMISSING: rtl/pulpissimo/pulpissimo.sv:385: Cell has missing pin: ref_clk_o
%Warning-PINMISSING: Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: rtl/pulpissimo/pulpissimo.sv:385: Cell has missing pin: rstn_o
%Warning-PINMISSING: ips/pulp_soc/rtl/pulp_soc//pulp_soc.sv:657: Cell has missing pin: zynq_soc_clk_i
%Warning-PINMISSING: ips/pulp_soc/rtl/pulp_soc//pulp_soc.sv:657: Cell has missing pin: zynq_per_clk_i
%Warning-REDEFMACRO: rtl/includes/periph_bus_defines.sv:19: Redefining existing define: NB_MASTER, with different value: 11
%Warning-REDEFMACRO: rtl/includes/soc_bus_defines.sv:116: Previous definition is here, with value: 4
%Warning-STMTDLY: ips/pulp_soc/rtl/components//freq_meter.sv:36: Unsupported: Ignoring delay on this delayed statement.
%Error: rtl/pulpissimo//soc_domain.sv:282: Pin not found: zynq_soc_clk_i
%Error: rtl/pulpissimo//soc_domain.sv:283: Pin not found: zynq_per_clk_i
%Warning-IMPLICIT: rtl/pulpissimo/pulpissimo.sv:633: Signal definition not found, creating implicitly: s_safe2soc_spi_master0_mode
%Warning-IMPLICIT: rtl/pulpissimo/pulpissimo.sv:782: Signal definition not found, creating implicitly: s_boot_l2
%Error: rtl/pulpissimo/pulpissimo.sv:809: Pin not found: zynq_cluster_clk_i
%Error: rtl/pulpissimo/pulpissimo.sv:800: Can't find definition of variable: zynq_per_clk_i
%Warning-IMPLICIT: rtl/pulpissimo/pulpissimo.sv:1029: Signal definition not found, creating implicitly: s_so
%Error: Exiting due to 4 error(s), 10 warning(s)
%Error: Command Failed /usr/local/bin/verilator_bin -DPULP_FPGA_EMUL --error-limit 100 -cc ./rtl/pulpissimo/pulpissimo.sv -I./rtl/pulpissimo/ -I./rtl/includes -I./ips/jtag_pulp/src/ -I./ips/pulp_soc/rtl/pulp_soc/ -I./ips/pulp_soc/rtl/ -I./ips/pulp_soc/rtl/components/ -I./ips/axi/axi_slice_dc/ -I./ips/common_cells/ -I./ips/L2_tcdm_hybrid_interco/TB/ -I./ips/L2_tcdm_hybrid_interco/RTL/XBAR_L2/ -I./ips/L2_tcdm_hybrid_interco/RTL/XBAR_BRIDGE/ -I./ips/L2_tcdm_hybrid_interco/RTL/ -I./ips/L2_tcdm_hybrid_interco/RTL/axi_2_lint/ -I./ips/axi/axi_slice/ -I./ips/timer_unit/ -I./ips/apb/apb_node/ -I./ips/adv_dbg_if/rtl/
`
but
`pulpissimo$ grep "zynq_soc_clk_i" ./rtl/pulpissimo/soc_domain.sv
    input  logic                             zynq_soc_clk_i,
        .zynq_soc_clk_i               ( zynq_soc_clk_i               ),
`
and
`pulpissimo$ grep "zynq_per_clk_i" ./rtl/pulpissimo/soc_domain.sv
    input  logic                             zynq_per_clk_i,
        .zynq_per_clk_i               ( zynq_per_clk_i               ),
`
`pulpissimo$ verilator --version
Verilator 3.924 2018-06-12 rev UNKNOWN_REV`
Could you help me?
Thanks.

meggiman, 2021-02-02:
if this is still relevant, please re-open another issue

closed 2021-02-02 by meggiman