pulp-platform/pulpissimo #28: saxpy with FPU no floating instruction.

lfernand4, 2018-07-20:
I want to compare the benefit of the FPU. But for the moment I see that the RISCY+FPU is slower.
I use the following function : 
[image]

x and y are float and the coef is 1.2. 

The result : 
[image]

I use this command to disassemble: make dis disopt="-d -l" | less

And I see that RISCY+FPU don't use floating instruction : 

[image]

Do you know why ?

Thanks in advance.

haugoug, 2018-08-06:
It seems you are using double not float, and double are SW emulated. Make sure all constants are described like 1.5f. 
Also did you use this config: configs/pulpissimo.sh
Could you show one of the gcc command which was executed to compile your code.

lfernand4, 2018-08-16:
thanks  ! it work with PULPISSIMO. 
But with PULP it doesn't work in float if I define ceof in float (1.2f).

closed 2019-04-25 by lfernand4