pulp-platform/pulpissimo #31: where can i get the source code of boot_code.cde ?

LGTMCU, 2018-09-18:
thanks a lot?

FrancescoConti, 2018-09-18:
I'm not sure it's part of the open-source release. @haugoug do you know?

haugoug, 2018-09-18:
This is indeed not yet open-source. Moreoever this cde file is actually from another chip, we should have removed it. I will try to regenerate a good cde file and open-source the boot code, I keep this tracker open and let you know.

haugoug, 2018-09-18:
You can now get the boot code here: https://github.com/pulp-platform/boot-code
I've also updated the cde file in pulpissimo.
The boot code is now compiled with the SDK if you take the latest from master. You can convert it to cde file and import it to pulpissimo sources by executing "make import_bootcode" from pulpissimo top directory. You will then have to recompile the rtl platform.
Once you want to boot from flash, you have to first configure the test with:
make conf boot=rom
Then compile and run:
make all run
if you look at the instruction traces, you should now see the core starting at 0x1a000000 instead of 0x1c000000. You should also see some activity on spi pads.

closed 2018-09-18 by haugoug