pulp-platform/pulpissimo #32: use verilog files to run modelsim simulation

sharinoz, 2018-11-14:
hi all,
we are currently in the middle of a project with involves using the Modelsim to run a pulpisimo simulation in it, and I am trying to get the .v and .sv files that are necessary for it but i am not sure if i have all the files i need,
does any one happened to work those two together and can guide me which files i need to run the modelsim simulation?

thanks

FrancescoConti, 2018-11-14:
Hi @sharinoz , it is not clear to me what is not working. Are you following the README? https://github.com/pulp-platform/pulpissimo/blob/master/README.md#building-the-rtl-simulation-platform

sharinoz, 2018-11-14:
i want to run a pulpisimo simulation (as basic as can it be, without any unnecessary extensions, etc..) on modelsim.
i didn't saw any reference for it in the README

FrancescoConti, 2018-11-14:
There is a whole "Getting started" section at the link I pointed you to...

sharinoz, 2018-11-14:
hi, i am trying to follow the README but i get stuck at that point:

# 10.6c 

  

# source /home/rotemshahar/pulpissimo/sim/tcl_files/config/run_and_exit.tcl 

# source /home/rotemshahar/pulpissimo/sim/tcl_files/run.tcl 

# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/home/rotemshahar/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf "+VSIM_PATH=/home/rotemshahar/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gENABLE_DEV_DPI=0 -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 

# Start time: 19:06:44 on Nov 14,2018 

# ** Error (suppressible): (vsim-19) Failed to access library 'work' at "work". 

# No such file or directory. (errno = ENOENT) 

# Error loading design 

Error loading design 

# End time: 19:06:44 on Nov 14,2018, Elapsed time: 0:00:00 

# Errors: 1, Warnings: 0 

VSIM reported an error, leaving 

/home/rotemshahar/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/__rules.mk:143: recipe for target 'run' failed 

make: *** [run] Error 1

FrancescoConti, 2018-11-14:
Can you post the output of the "Building the RTL simulation platform" section?

Sati2001, 2020-05-18:
I have the same problem 
Error loading design
Please anyone help me

bluewww, 2020-05-18:
The error looks to me like the rtl platform was not compiled.

Maybe forgot the `make build` part?

closed 2020-11-17 by bluewww