pulp-platform/pulpissimo #33: How to port the freertos to pulpissimo from pulpinfo platform?

shiner-chen, 2018-11-19:
We know the pulpinfo platform support freertos and I tried it. Now I want to port  the freertos to Pulpissimo. Because the freertos depends on string_lib and sys_lib, I copied the sys_lib and string_lib directories into  pulp-rt-examples and  modified the Makefile as below to  build the sys_lib on pulpissimo, but failed! How to write the Makefile to build the static library in pulpissimo?

PULP_LIBS = sys_lib
PULP_APP_SRCS = src/exceptions.c src/gpio.c src/int.c src/spi.c src/timer.c src/uart.c src/utils.c src/i2c.c

PULP_CFLAGS += -I$(CURDIR)/inc -I$(CURDIR)/../string_lib/inc 
#PULP_CFLAGS += -O3 -falign-functions=16 -funroll-all-loops
PULP_CFLAGS += -Os -g -fno-jump-tables -Werror

INSTALL_FILES += $(shell find inc -name *.h)

include $(PULP_SDK_HOME)/install/rules/pulp_rt.mk

error log:

make all
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/exceptions.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/exceptions.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/gpio.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/gpio.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/int.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/int.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/spi.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/spi.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/timer.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/timer.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/uart.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/uart.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/utils.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/utils.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c src/i2c.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc/src/i2c.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/fc_config.h    -MMD -MP -c /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/rt_conf.c -o /home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/sys_lib/fc//home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/rt_conf.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imcxpulpv2 -D__riscv__ -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/inc -I/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/../string_lib/inc  -Os -g -fno-jump-tables -Werror  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulpissimo/pulp-sdk/pk [... cut ...]
  File "/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/bin/pulp-run", line 48, in <module>
    retval = runner.run()
  File "/home/chenx/build/pulpissimo/pulp-sdk/pkg/sdk/dev/install/ws/python/plp_runner.py", line 185, in run
    self.system_tree = plptree.get_configs_from_file(self.config.getOption('configFile'))[0]
IndexError: list index out of range
/home/chenx/build/pulpissimo/examples/pulp-rt-examples/libs/sys_lib/build/pulpissimo-riscy/__rules.mk:107: recipe for target 'prepare' failed
make: *** [prepare] Error 1

shiner-chen, 2018-11-19:
Worked around the above issue by just run "make build" command to build the libraries, but when run the freertos in modelsim, I get the the follwoing error!
Launching VSIM with command:
export VSIM_RUNNER_FLAGS="-gLOAD_L2=JTAG -permit_unmatched_virtual_intf" && vsim -64 -c -do 'source /home/chenx/build/pulpissimo/sim/tcl_files/config/run_and_exit.tcl' -do 'source /home/chenx/build/pulpissimo/sim/tcl_files/run.tcl; run_and_exit;'
Reading pref.tcl

# 10.4c

# source /home/chenx/build/pulpissimo/sim/tcl_files/config/run_and_exit.tcl
# source /home/chenx/build/pulpissimo/sim/tcl_files/run.tcl
# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/home/chenx/build/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf "+VSIM_PATH=/home/chenx/build/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gENABLE_DEV_DPI=0 -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 
# Start time: 17:26:59 on Nov 19,2018
# //  Questa Sim-64
# //  Version 10.4c linux_x86_64 Jul 19 2015
# //
# //  Copyright 1991-2015 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //  THIS DOCUMENT CONTAINS TRADE SECRETS AND COMMERCIAL OR FINANCIAL
# //  INFORMATION THAT ARE PRIVILEGED, CONFIDENTIAL, AND EXEMPT FROM
# //  DISCLOSURE UNDER THE FREEDOM OF INFORMATION ACT, 5 U.S.C. SECTION 552.
# //  FURTHERMORE, THIS INFORMATION IS PROHIBITED FROM DISCLOSURE UNDER
# //  THE TRADE SECRETS ACT, 18 U.S.C. SECTION 1905.
# //
# ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_soc/i_FLL_digital/TQ has no driver.
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_soc/i_FLL_digital/JTQ has no driver.
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_per/i_FLL_digital/TQ has no driver.
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_per/i_FLL_digital/JTQ has no driver.
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_cluster/i_FLL_digital/TQ has no driver.
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_cluster/i_FLL_digital/JTQ has no driver.
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-3040) Command line generic/parameter "USE_SDVT_I2S" not found in design.
# ** Warning: (vsim-3040) Command line generic/parameter "USE_SDVT_SPI" not found in design.
#  run_and_exit
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "stdout/uart" for writing.
# No such file or directory. (errno = ENOENT)    : /home/chenx/build/pulpissimo/sim/../rtl/vip/uart_tb_rx.sv(44)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/genblk4/i_rx_mod
# START_ADDR[0] = 0x1a100000; END_ADDR[0] = 0X1a200000
# START_ADDR[1] = 0x10000000; END_ADDR[1] = 0X10400000
# [TB]                    0 - Using FLL
# [TB]                    0 - Not using CAM SDVT
# [TB]                    0 - Asserting hard reset
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "stdout/stdout_fake_pe0_0" for writing.
# No such file or directory. (errno = ENOENT)    : /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/STDOUT_PRINTF_CLUSTER
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_0_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_CLUSTER
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "stdout/stdout_fake_pe31_0" for writing.
# No such file or directory. (errno = ENOENT)    : /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/STDOUT_PRINTF_FC
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_31_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_FC
# ** Warning: (vsim-8315) /home/chenx/build/pulpissimo/sim/../ips/pulp_soc/rtl/fc/fc_demux.sv(97): No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 2  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/fc_demux_instr_i
# ** Warning: (vsim-8315) /home/chenx/build/pulpissimo/sim/../ips/pulp_soc/rtl/fc/fc_demux.sv(97): No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 2  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/fc_demux_data_i
# JTAG: Bypass Test Passed
# JTAG: Tap ID: 149511c3
# [TB]             47401000 - Virtually removing ROM as JTAG boot was selected in testbench
# [test_mode_if]             48401000 - Init
[... cut ...]
# ** Note: $stop    : /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_pulp.sv(679)
#    Time: 18471773188 ps  Iteration: 0  Instance: /tb_pulp
# Break at /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 679
# Stopped at /home/chenx/build/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 679
# End time: 17:27:30 on Nov 19,2018, Elapsed time: 0:00:31
# Errors: 0, Warnings: 15
VSIM reported an error, leaving
/home/chenx/build/pulpissimo/examples/pulp-rt-examples/freertos/build/pulpissimo-riscy/__rules.mk:137: recipe for target 'run' failed
make: *** [run] Error 1

haugoug, 2019-02-11:
You should try with this version instead:
https://github.com/GreenWaves-Technologies/freeRTOS
It has been ported to Gap chip which is derived from Pulp and should be much easier to port to pulpissimo.

closed 2019-02-11 by haugoug