pulp-platform/pulpissimo #36: Does pulpissimo support burn the application into flash by JTAG?

shiner-chen, 2018-12-28:
Our FPGA engineer port the pulpissimo code to Xilinx FPGA board. We have verified the peripherals, such as I2C, UART, SPIM, Camera, I2S. Now we want to debug the board boot from external nor flash, we want to implement it by the following steps:
1. Modify the bootrom code to load the application from flash and run it.
2. Burn the application binary into flash through JTAG.

We checked the pulp-debug-bridge code, it seems that it does not support burn app binary into flash. So we have two options:
1. Modify the pulp-debug-bridge tools and JTAG RTL code to support burn app
2. Maybe use the OpenOCD tool + modification of JTAG RTL to implement it, because openocd has flash program interface.

Which way is less work? If any other better way, please info us, thanks a lot!

daviderossi1982, 2018-12-28:
We are working on the porting of RISC-V compliant specs to PULPissimo & PULP, full OpenOCD tool support will come with this. I think the best solution is to wait for the new PULP debug to be implemented, and then use this as a starting point in case further changes are required.

daviderossi1982, 2018-12-28:
it is under development, not integrated in the master branch yet

shiner-chen, 2018-12-29:
@daviderossi1982 What's the plan for integrating new PULP debug interface into master branch? Have any plan to support the flash write_image command from OpenOCD? Thanks.

shiner-chen, 2019-01-16:
@daviderossi1982  what's the answer for above question? thanks a lot.

daviderossi1982, 2019-01-16:
Sorry I have missed it. The plan is to have it integrated in the master branch by Q2 2019. No plans to support the flash write_image command from OpenOCD, at least at the moment.

closed 2019-02-11 by haugoug