pulp-platform/pulpissimo #37: edge syntax error while building the RTL simulation platform

ido-raz, 2019-01-01:
Hello,

I am trying to build the simulation environment of the PULPissimo and follow the instruction in the README.
When running the make command in the 'Building the RTL simulation platform' section I get the following error when the make get to build the 'tbtools' (the same error happened when the make tries to build the 'tb' component):
(Also attached print screen)
  Building tbtools/tbtools
** Warning: (vlog-52) Unknown message number: 13314.
** Error: ../ips/tbtools/dpi_models/dpi_models.sv(361): near "edge": syntax error, unexpected edge
** Error: ../ips/tbtools/dpi_models/dpi_models.sv(394): near "edge": syntax error, unexpected edge
** Error: ../ips/tbtools/dpi_models/dpi_models.sv(440): near "edge": syntax error, unexpected edge
** Error: ../ips/tbtools/tb_driver/tb_driver.sv(128): (vlog-2164) Class or package 'dpi_models' not found.
** Error: ../ips/tbtools/tb_driver/tb_driver.sv(128): Undefined variable: dpi_models.
** Error: ../ips/tbtools/tb_driver/tb_driver.sv(128): near "::": syntax error, unexpected ::
** Error: ../ips/tbtools/tb_driver/tb_driver.sv(131): Illegal declaration after the statement near line '128'.  Declarations must precede statements.  Look for stray semicolons.
make[2]: *** [/data.cc/data/a/home/cc/students/enginer/idoraz/pulp_sim/pulpissimo/sim/modelsim_libs/tbtools_lib/tbtools.vmake] Error 2
make[1]: *** [build] Error 2
make: *** [build] Error 2
[image]

Please help.
Thanks,
Ido

FrancescoConti, 2019-01-08:
Hi @ido-raz sorry for not answering earlier -- this came in during the winter holidays. 
Can you report the output of the `vlog -version` command?

ido-raz, 2019-01-08:
Hi, Thank you for you response.
The output of the command vlog -version is:
QuestaSim-64 vlog 10.0c Compiler 2011.07 Jul 21 2011

ido-raz, 2019-01-14:
Is this info gives any clue about the reason to the error?
Thanks

Almidi, 2019-07-16:
Hello
Same problem here
vlog -version is:
Model Technology ModelSim ALTERA vlog 10.1d Compiler 2012.11 Nov  2 2012

bluewww, 2019-07-16:
Your version of modelsim doesn't support these SystemVerilog features. `PULPissimo` itself doesn't use most of `common_cells` though and uses a more conservative set of features from SystemVerilog so I suppose It would be possible to get it running by removing the the offending stuff from `common_cells`.

I wen through all our modelsim/questasim versions and it seems like
`Questa Sim  vsim 10.3e Simulator 2015.02 Feb 23 2015` is at least required for now.

closed 2021-02-02 by meggiman