pulp-platform/pulpissimo #42: hwme/hwme.c:38: undefined reference to `hwme_vectstride2_set'

tangshibin, 2019-02-13:
I want try the MAC-accelerator;
The sdk is compiled following  the instructions: https://github.com/pulp-platform/pulp-sdk;
the toolchain is compiled following the instructions: https://github.com/pulp-platform/pulp-riscv-gnu-toolchain;

But I got the error as following:
pulp-sdk/pulp-rt-examples/accelerators/hwme/hwme.c:38: undefined reference to `hwme_vectstride2_set'
If it is commented in hwme.c, the hwme.c can be compiled successfully.

FrancescoConti, 2019-02-13:
@haugoug I moved this to the PULPissimo tracker. Then I’ll deal with it myself ASAP

bluewww, 2021-02-02:
This is a pulp-sdk bug which is probably not relevant anymore with the new pulp-sdk changes. Please open another issue if this is still a problem.

closed 2021-02-02 by bluewww