pulp-platform/pulpissimo #43: pulp_rt_examples/hello,  how to run it on GVSOC

tangshibin, 2019-02-14:
Hello, I want to try the pulp_rt_examples on GVSOC, which is the default platform described in Builder DOC.
But it always start the RTL simulation. Because I haven't install ModelSim, I got the error vsim command is not found.

If I want to run all pulp_rt_exmaples with GVSOC, which steps are needed to modified the Makefile?

FrancescoConti, 2019-02-14:
When you source the SDK, you should configure it for the correct target and simulation platform, e.g. (i go by memory as I am on mobile)
```
. pkg/sdk/config/pulpissimo.sh
. pkg/sdk/config/platform-gvsoc.sh
```

tangshibin, 2019-02-14:
hello, 
what's the difference between pulp-builder and pulp-sdk?
The error about "hello-example and GVSOC" occurs in pulp-builder, and it is ok in pulp-sdk.
But I run pulp-rt-example/hwme on GVSOC, the error is got as following:
/*******************************************************************************
2546704416: 65883: [/sys/board/chip/soc/fc/warning                   ] Invalid access (offset: 0x1a10c004, size: 0x4, is_write: 0)
********************************************************************************/

tangshibin, 2019-02-14:
pkg/sdk/config/platform-gvsoc.sh is not existed in pulp-builder/configs/

haugoug, 2019-02-14:
pulp-builder is a simplified build process to more easily get the runtime for the RTL platform. To use gvsoc, you have to use pulp-sdk.

closed 2019-02-20 by FrancescoConti

Ahmed1Zaky, 2019-04-16:
> hello,
> what's the difference between pulp-builder and pulp-sdk?
> The error about "hello-example and GVSOC" occurs in pulp-builder, and it is ok in pulp-sdk.
> But I run pulp-rt-example/hwme on GVSOC, [...]

Have you found a solution for this error ? I am facing the same now.