pulp-platform/pulpissimo #44: Error while simulating matrix multiplication

tangshibin, 2019-02-20:
I writing an matrix multiplication and use the Makefile in hello;
I got an error as in simulation, it is 
[image]

haugoug, 2019-02-20:
This usually happens when you use an uninitialized variable. You can have a look in the file trace_core_1f_0.log, this contains the list of executed instructions, you should see some XX

tangshibin, 2019-02-21:
I have checked the trace_core_1f_0.log； But there is no instructions in it; 
I also chech the trace_core_1f_0.log in other examples, It contains the traces;

haugoug, 2019-02-21:
It could be that the file is not flushed, make sure you properly stop the platform so that it happens. Otherwise, you'll have to look on the signals to see what the core is doing

tangshibin, 2019-02-21:
I have trace the program and found that the error is reported from the function "malloc".
Does the malloc is tested on the platform?

haugoug, 2019-02-21:
There is no malloc available (see the other tracker)