pulp-platform/pulpissimo #46: FFT error on RTL simulation platform

tangshibin, 2019-02-23:
hi ,
 I try to simulate a FFT program on pulpissimo.
 I have verify the FFT program on GVSOC platform,  which has gotten the successful results.
 But when I simulating the same program on RTL platform, I got the following error:
 ===========================================
# [dbg_if_soc] WRITE32 burst @1c013400 for        1024 bytes.
# [dbg_if_soc] WRITE32 burst @1c013800 for        1024 bytes.
# [dbg_if_soc] WRITE32 burst @1c013c00 for        1024 bytes.
# [dbg_if_soc] WRITE32 burst @1c014000 for        1024 bytes.
# [dbg_if_soc] WRITE32 burst @1c014400 for         288 bytes.
# [TB]          68436173188 - Triggering fetch enable
# [TRACER] Output filename is: trace_core_1f_0.log
#          68446881860: Illegal instruction (core 0) at PC 0x1c008080:                                                                                                                        
#          68447194436, Branch decision is X in module tb_pulp.i_dut.soc_domain_i.pulp_soc_i.fc_subsystem_i.FC_CORE.TB_RISCV.lFC_CORE.RISCV_CORE.id_stage_i
# ** Note: $stop    : /home/lgq/tangshibin/riscv2019/pulpissimo/sim/../ips/riscv/./rtl/riscv_id_stage.sv(1558)
#    Time: 68447194436 ps  Iteration: 19  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/TB_RISCV/lFC_CORE/RISCV_CORE/id_stage_i
# Break at /home/lgq/tangshibin/riscv2019/pulpissimo/sim/../ips/riscv/./rtl/riscv_id_stage.sv line 1558
============================================
I also check the wave with modelsim. PC: 1c008080 is the first instruction for RTL simulation;
But in GVSOC, the first instruction is PC:1a000080. I don't know why?

haugoug, 2019-02-23:
On gvsoc, the execution starts from the rom at 1a000080, which then load the program from spi flash, while on the RTL, the program is loaded through JTAG and directly starts at 0x1c008080.
We are currently updating the core, which is breaking the compatibility, maybe you get something wrong due to that.
Could you attach your ELF binary (the one normally in build/pulpissimo/test/test) and also gives the git hash tags your are using for the pulpissimo RTL, and also for the sdk ?

tangshibin, 2019-02-23:
The hash code is 
            sdk: 8e35524f348500694c5a24cba36254e84cd0df08
pulpissimo: 2f0a13ca34d621b467961bbbb41bfe1e6713138f

ELF binary is 
[fft.zip](https://github.com/pulp-platform/pulpissimo/files/2896612/fft.zip)

tangshibin, 2019-02-23:
I aslo compare the hello and FFT on the RTL simulating platform;
For hello, the first instruction is 0x0200006f, which can be found in fft.bin;
But for FFT, the first instruction is 0x7d59ca9f, which can not be found in fft.bin

haugoug, 2019-02-23:
The problem is a limitation of the RTL platform. When booting from JTAG, the binary entry is hard-coded to 0x1c008080 (line 631 in rtl/tb/tb_pulp.sv). On Pulpissimo data is put between 0x1c000000 and 0x1c008000 and code is put starting at 0x1c008000. The linker script is made in a way that if the data is bigger than 0x8000, the code is shifted, which also shifts the entry point and make it fails on the RTL platform. As you have a lot of data in your binary, this is what happens.
There are 2 solutions:
- You can modify the testbench to put your entry point 0x1c00ec80 but you'll have to do it everytime your binary is different
- You can boot from spiflash. You can follow the instructions to install the model and then compile and run your application with this command:
make conf boot=rom all run CONFIG_OPT=vsim/boot-mode=rom

tangshibin, 2019-02-23:
Thanks for your reply；
I try the two solutions;
First, it works;
Second, it still reports error at the first insturction simulation.

tangshibin, 2019-02-25:
In FFT test program, I prepared input (2*80*2) and output(2*80*1) to verification the compuation.
When printf the results in GVSOC platform, I found that the output data is not the same as I prepared.  The computation results with input is right.
Is it related to the linkder script?

tangshibin, 2019-02-25:
Does the whole L2 (512KB) can be used to save the input program?

haugoug, 2019-02-27:
Yes the whole L2 can be used to load the input program