pulp-platform/pulpissimo #49: pulp-rt-examples/periph/uart/loopback not working with pulpissimo

Zashas, 2019-03-06:
Hi @haugoug and @politicante,
I'm trying to run the example script https://github.com/pulp-platform/pulp-rt-examples/tree/master/periph/uart/loopback with pulpissimo and the default testbench (which sets uart TX = RX), however it is not able to finish. The following call is perpetually blocking : `rt_uart_write(uart, tx_buffer, BUFFER_SIZE, NULL);`. 

More specifically, it is the `__rt_wait_event_check(event, call_event);` child call performed in `rt_periph_copy` that seems to block everything. I didn't have the time to dive into the details of the Events Unit and how it interacts with udma, but my intuition here is that the processor is not able to properly interact with udma, and nothing is sent on the UART TX. As rt_uart_write in this case is blocking and udma is idle, no event is sent back to the CPU although the scheduler expects one.

Here below is a QuestaSim screenshot showing the whole RTL simulation, we can see that: 

- The CPU tries to communicate with udma (cf. APB signals)
- Despite the DMA commands, the udma UART TX interface is not starting a transfer, the FSM stays idle (cf. CS signal)

[image]

Did I misunderstand something in the setup, or would this be a bug ? Any help would be much appreciated. Thanks.

Zashas, 2019-03-07:
OK mea culpa. I did not notice the README update (2f0a13ca34d621b467961bbbb41bfe1e6713138f) which fixes a specific version of the SDK, and I naively pulled both repos to master yesterday. It indeed works fine with the given SDK build.

It could maybe be interesting to create a stable-pulpissimo branch in pulp-builder, which would guarantee a stable SDK ?

closed 2019-03-07 by Zashas

Shiva-prog123, 2020-09-05:
@Zashas I am alos trying to run the same code....but my c to hex conversion giving error..How you run the above UART code?