pulp-platform/pulpissimo #55: pulp-soc syntax error

MikkeNaa, 2019-05-27:
When I try to synthesize Pulpissimo (design compiler L2016-03), error comes from pulp_soc module:

Information: Building the design 'pulp_soc' instantiated from design 'soc_domain_0_1_32_64_32_6_4_6_8_4_8_8' with
	the parameters "CORE_TYPE=0,USE_FPU=1,AXI_ADDR_WIDTH=32,AXI_DATA_IN_WIDTH=64,AXI_DATA_OUT_WIDTH=32,AXI_ID_IN_WIDTH=6,AXI_ID_OUT_WIDTH=4,AXI_USER_WIDTH=6,EVNT_WIDTH=8,BUFFER_WIDTH=8". (HDL-193)
Error:  ../pulpissimo/ips/pulp_soc/rtl/pulp_soc/pulp_soc.sv:269: Syntax error at or near token 'FC_Core_MHARTID': arrays do not have named members. (VER-294)

Line 269: localparam dm::hartinfo_t [NrHarts-1:0] HARTINFO = '{FC_Core_MHARTID: RI5CY_HARTINFO, default: 0};

This issue is relatively new as this code didn't exist May 6th!

FrancescoConti, 2019-05-27:
Mmm, this issue seems associated with some relatively "advanced" SystemVerilog code, I think it's likely that a newer version of Design Compiler might work. Here it seems it doesn't recognize that each element of `HARTINFO` (array of structs) should be assigned the value `{FC_Core_MHARTID: RI5CY_HARTINFO, default: 0}`.

Maybe @bluewww who introduced this can help you better...

bluewww, 2019-05-27:
Ok I will look into it, I think someone in our group already had to change that line for the FPGA port.

MikkeNaa, 2019-05-28:
I changed to DC 2018.06-SP4. Still same issue.

meggiman, 2019-06-20:
Hi @MikkeNaa can you try out the new bugfix_i55 branch I just created on the pulp_soc repository? 
I changed the offending line to use some less modern language construct. The same change was necessary for Vivado so it might also do the trick for Synopsys DC.

bluewww, 2019-08-05:
Fixed https://github.com/pulp-platform/pulp_soc/pull/7. Please let me now if it works for you.

MikkeNaa, 2019-08-16:
I confirm that it works now

closed 2019-08-16 by MikkeNaa