pulp-platform/pulpissimo #60: The OpenOCD error on RTL simulation platform

shiner-chen, 2019-07-15:
Following the guide:https://github.com/pulp-platform/pulpissimo/blob/master/rtl/tb/README.md, I tried the OpenOCD to connect the RTL platform and get the errors below.
on RTL platform side, continuously print the following log:
```
Write 1 0 0
Tick with: TCK=1 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 2
Write 0 1 0
Tick with: TCK=0 TMS=1 TDI=0 TRSTn=1 --> TDO=0
Received a command R
Read req
Tick with: TCK=0 TMS=1 TDI=0 TRSTn=1 --> TDO=0
Received a command 6
Write 1 1 0
Tick with: TCK=1 TMS=1 TDI=0 TRSTn=1 --> TDO=0
Received a command 2
Write 0 1 0
Tick with: TCK=0 TMS=1 TDI=0 TRSTn=1 --> TDO=0
Received a command 6
Write 1 1 0
Tick with: TCK=1 TMS=1 TDI=0 TRSTn=1 --> TDO=0
Received a command 0
Write 0 0 0
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 4
Write 1 0 0
Tick with: TCK=1 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 0
Write 0 0 0
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 0
Write 0 0 0
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 4
Write 1 0 0
Tick with: TCK=1 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 0
Write 0 0 0
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 4
Write 1 0 0
Tick with: TCK=1 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 0
Write 0 0 0
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 4
Write 1 0 0
Tick with: TCK=1 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command 0
Write 0 0 0
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command b
blink off
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Received a command Q
Quit req
Remote end disconnected
Tick with: TCK=0 TMS=0 TDI=0 TRSTn=1 --> TDO=0
Attempting to accept client socket
```
The OpenOCD side:
```
chenx@wangzhongping-All-Series:~/build/pulpissimo/rtl/tb$ /opt/openocd/bin/openocd -f pulpissimo_debug.cfg
Open On-Chip Debugger 0.10.0+dev-00529-g9fac2de (2019-07-11-17:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 32 command.c:544 command_print(): debug_level: 4
Debug: 14 32 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_khz 1000
Debug: 15 32 command.c:143 script_debug(): command - adapter_khz ocd_adapter_khz 1000
Debug: 17 32 core.c:1645 jtag_config_khz(): handle jtag khz
[... 269 lines cut ...]
Debug: 297 717361 tcl.c:1225 handle_flash_init_command(): Initializing flash devices...
Debug: 298 717361 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_mflash init
Debug: 299 717361 command.c:143 script_debug(): command - ocd_mflash ocd_mflash init
Debug: 301 717361 mflash.c:1377 handle_mflash_init_command(): Initializing mflash devices...
Debug: 302 717361 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_nand init
Debug: 303 717361 command.c:143 script_debug(): command - ocd_nand ocd_nand init
Debug: 305 717361 tcl.c:497 handle_nand_init_command(): Initializing NAND devices...
Debug: 306 717361 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_pld init
Debug: 307 717361 command.c:143 script_debug(): command - ocd_pld ocd_pld init
Debug: 309 717361 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 310 717361 gdb_server.c:3412 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 311 717361 server.c:311 add_service(): Listening on port 3333 for gdb connections
Debug: 312 717361 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_halt
Debug: 313 717361 command.c:143 script_debug(): command - halt ocd_halt
Debug: 315 717361 target.c:3041 handle_halt_command(): -
Error: 316 717361 target.c:572 target_halt(): Target not examined yet
Debug: 317 717361 command.c:651 run_command(): Command 'halt' failed with error code -4
User : 318 717361 command.c:715 command_run_line(): 
Debug: 319 717361 riscv.c:276 riscv_deinit_target(): riscv_deinit_target()
Debug: 320 717361 riscv-013.c:1359 deinit_target(): riscv_deinit_target()
Debug: 321 717361 target.c:1941 target_free_all_working_areas_restore(): freeing all working areas
Info : 322 717361 remote_bitbang.c:120 remote_bitbang_quit(): remote_bitbang interface quit
chenx@wangzhongping-All-Series:~/build/pulpissimo/rtl/tb$ 
```
It seems that the RTL platform does not support the target_halt()? How to resolve it ?

bluewww, 2019-07-15:
```
Info : 288 717361 riscv-013.c:1473 examine(): datacount=2 progbufsize=8
Debug: 289 717361 riscv-013.c:1544 examine(): Enumerated 0 harts
Error: 290 717361 riscv-013.c:1547 examine(): No harts found!
```
RISC-V OpenOCD assumes at most 32 harts exists and that the harts are numbered 0 to 31. Currently PULPissimo uses the hartid 992 which is totally conformant to the debug specification but not supported out of the box. I raised the issue on riscv-openocd but there is no real solution yet.

A hacky workaround is compiling riscv-openocd with the MAX_HARTID macro set to 1024 or more

shiner-chen, 2019-07-16:
@bluewww  Thank you for your reply,  I tried the patch for  openocd, but  openocd still cannot work. It entered the loop for scan all hartid, because of the dm->hart_count < 0. please see the code below in riscv-013.c
```
 LOG_ERROR("hart count =%d\n", dm->hart_count);  //Added by me to display the hart count
1555     /* Before doing anything else we must first enumerate the harts. */
1556     if (dm->hart_count < 0) {
1557         for (int i = 0; i < MIN(RISCV_MAX_HARTS, 1 << info->hartsellen); ++i) {
1558             r->current_hartid = i;
1559             if (riscv013_select_current_hart(target) != ERROR_OK)
1560                 return ERROR_FAIL;
1561
1562             uint32_t s;
1563             if (dmstatus_read(target, &s, true) != ERROR_OK)
1564                 return ERROR_FAIL;
1565             if (get_field(s, DMI_DMSTATUS_ANYNONEXISTENT))
1566                 break;
1567             dm->hart_count = i + 1;
1568
1569             if (get_field(s, DMI_DMSTATUS_ANYHAVERESET))
1570                 dmi_write(target, DMI_DMCONTROL,
1571                         set_hartsel(DMI_DMCONTROL_DMACTIVE | DMI_DMCONTROL_ACKHAVERESET, i));
1572         }
1573
1574         LOG_DEBUG("Detected %d harts.", dm->hart_count);
1575     }
```
The following log is outputted by openocd.
``` 
Debug: 232 240825 core.c:1190 jtag_validate_ircapture(): IR capture validation scan
Debug: 233 249192 core.c:1248 jtag_validate_ircapture(): riscv.unknown0: IR capture 0x05
Debug: 234 249192 core.c:1248 jtag_validate_ircapture(): riscv.cpu: IR capture 0x05
Debug: 235 249192 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_dap init
Debug: 236 249192 command.c:143 script_debug(): command - ocd_dap ocd_dap init
Debug: 237 249192 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (249192). This may cause trouble with GDB connections.
Debug: 240 249192 arm_dap.c:105 dap_init_all(): Initializing all DAPs ...
Debug: 241 249192 openocd.c:159 handle_init_command(): Examining targets...
Debug: 242 249192 target.c:1609 target_call_event_callbacks(): target event 17 (examine-start) for core 992
Debug: 243 249192 riscv.c:916 riscv_examine(): riscv_examine()
Debug: 244 266484 riscv.c:356 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x1071
Debug: 245 266484 riscv.c:926 riscv_examine(): dtmcontrol=0x1071
Debug: 246 266484 riscv.c:928 riscv_examine():   version=0x1
Debug: 247 266484 riscv-013.c:1702 init_target(): init
Debug: 248 287699 riscv-013.c:448 dtmcontrol_scan(): DTMCS: 0x0 -> 0x1071
Debug: 249 287699 riscv-013.c:1447 examine(): dtmcontrol=0x1071
Debug: 250 287699 riscv-013.c:1448 examine():   dmireset=0
Debug: 251 287699 riscv-013.c:1449 examine():   idle=1
Debug: 252 287699 riscv-013.c:1450 examine():   dmistat=0
Debug: 253 287699 riscv-013.c:1451 examine():   abits=7
Debug: 254 287699 riscv-013.c:1452 examine():   version=1
Debug: 255 287699 riscv-013.c:259 get_dm(): [992] Allocating new DM
Debug: 256 312158 riscv-013.c:393 scan(): 41b 0i w 00000000 @10 -> + 00000000 @00
Debug: 257 328491 riscv-013.c:393 scan(): 41b 0i - 00000000 @10 -> + 00000000 @10
Debug: 258 349049 riscv-013.c:393 scan(): 41b 0i w 00000001 @10 -> + 00000000 @10
Debug: 259 349049 riscv-013.c:404 scan():  dmactive -> 
Debug: 260 365536 riscv-013.c:393 scan(): 41b 0i - 00000000 @10 -> + 00000001 @10
Debug: 261 365536 riscv-013.c:404 scan():  ->  dmactive
Debug: 262 385984 riscv-013.c:393 scan(): 41b 0i w 07ffffc1 @10 -> + 00000000 @10
Debug: 263 385984 riscv-013.c:404 scan():  hasel hartselhi=1023 hartsello=1023 dmactive -> 
Debug: 264 402421 riscv-013.c:393 scan(): 41b 0i - 00000000 @10 -> + 07ffffc1 @10
Debug: 265 402421 riscv-013.c:404 scan():  ->  hasel hartselhi=1023 hartsello=1023 dmactive
Debug: 266 422823 riscv-013.c:393 scan(): 41b 0i r 00000000 @10 -> + 00000000 @10
Debug: 267 439253 riscv-013.c:393 scan(): 41b 0i - 00000000 @10 -> b 00000000 @10
Debug: 268 439253 riscv-013.c:459 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=1, ac_busy_delay=0
Debug: 269 456520 riscv-013.c:448 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1c71
Debug: 270 476974 riscv-013.c:393 scan(): 41b 1i - 00000000 @10 -> + 03ffffc1 @10
Debug: 271 476974 riscv-013.c:404 scan():  ->  hartselhi=1023 hartsello=1023 dmactive
Debug: 272 497732 riscv-013.c:393 scan(): 41b 1i r 00000000 @11 -> + 00000000 @10
Debug: 273 514587 riscv-013.c:393 scan(): 41b 1i - 00000000 @11 -> b 00000000 @11
Debug: 274 514587 riscv-013.c:459 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=2, ac_busy_delay=0
Debug: 275 532096 riscv-013.c:448 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1071
Debug: 276 552520 riscv-013.c:393 scan(): 41b 2i - 00000000 @11 -> + 0000f082 @11
[... 47 lines cut ...]
# [CORE] Core settings: PULP_SECURE =           1, N_PMP_ENTRIES =          16, N_PMP_CFG           4
# ** Warning: (vsim-8315) /home/chenx/build/pulpissimo/sim/../ips/riscv/./rtl/riscv_pmp.sv(677): No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/lFC_CORE/RISCY_PMP/pmp_unit_i
# START_ADDR[0] = 0x1a100000; END_ADDR[0] = 0X1a400000
# START_ADDR[1] = 0x10000000; END_ADDR[1] = 0X10400000
# ** Warning: (vsim-8315) /home/chenx/build/pulpissimo/sim/../ips/pulp_soc/rtl/components/tcdm_arbiter_2x1.sv(45): No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/jtag_lint_arbiter_i
[... cut ...]
# ** Warning: (vsim-8315) /home/chenx/build/pulpissimo/sim/../ips/riscv/./rtl/riscv_pmp.sv(677): No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 2  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/lFC_CORE/RISCY_PMP/pmp_unit_i
# ** Warning: (vsim-8315) /home/chenx/build/pulpissimo/sim/../ips/pulp_soc/rtl/components/tcdm_arbiter_2x1.sv(45): No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 12  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/jtag_lint_arbiter_i
# [TB]       1ns - Releasing hard reset
Received a command 2
```

shiner-chen, 2019-07-17:
After I update to latest code of pulpissimo as below:
```
chenx@wangzhongping-All-Series:~/build/pulpissimo$ git log
commit eb4f4ee77e2d16354c0cc92c3dcee016f4f5e7fe
Merge: ff1c2c6 33edf56
Author: bluew <bluewww@users.noreply.github.com>
Date:   Wed Jul 10 19:36:42 2019 +0200

    Merge pull request #59 from pulp-platform/openocd
    
    OpenOCD

commit 33edf56a0599744d68aaa0f4bce00345d6699e2a
Author: bluew <bluewww@users.noreply.github.com>
Date:   Wed Jul 10 19:27:17 2019 +0200

    Import new bitbang server version
    
    This makes the server a bit more quiet when running, but the previous
    behaviour can be obtained by defining VERBOSE.
    
    Furthermore there is a small bug fix: When the remote hangs up we
    can get stuck in an infinite loop.

commit 275ea311ab47f835ce12522bf43063137b3aa501
Author: bluew <bluewww@users.noreply.github.com>
Date:   Wed Jul 10 19:26:43 2019 +0200

    Make toplevel makefile build the bitbang server. Update README

commit 7e894fb352d9ae0fad4c04982f958106a48c9eca
Author: bluew <bluewww@users.noreply.github.com>
Date:   Wed Jul 10 19:26:28 2019 +0200

    Add instruction for using openocd + tb

```
When run the hello demo on RTL platform, I got different error: "`Port number of out range: -265478504`" .
```
chenx@wangzhongping-All-Series:~/build/pulp-rt-examples/hello$ make conf CONFIG_OPT="vsim/tcl_args=-sv_lib vsim/tcl_args="${VSIM_PATH}"/../rtl/tb/remote_bitbang/librbs vsim/tcl_args=-gENABLE_OPENOCD=1" clean all run
plpflags gen --input=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config=platform=rtl  --config=vsim/tcl_args=-sv_lib  --config=vsim/tcl_args=/home/chenx/build/pulpissimo/sim/../rtl/tb/remote_bitbang/librbs  --config=vsim/tcl_args=-gENABLE_OPENOCD=1  --config=**/rt/type=pulp-rt --output-dir=/home/chenx/build/pulp-rt-examples/hello/build/pulpissimo --makefile=/home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/config.mk    --app=test 
plpconf --input=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config=platform=rtl  --config=vsim/tcl_args=-sv_lib  --config=vsim/tcl_args=/home/chenx/build/pulpissimo/sim/../rtl/tb/remote_bitbang/librbs  --config=vsim/tcl_args=-gENABLE_OPENOCD=1  --config=**/rt/type=pulp-rt --output=/home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/config.json 
rm -rf /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/test /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c test.c -o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_conf.c -o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/include -include /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.c -o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/test`
/opt/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/test /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/chenx/build/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  -nostartfiles -nostdlib -Wl,--gc-sections -L/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/rules -Tpulpissimo/link.ld -L/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/lib/pulpissimo -L/home/chenx/build/pulp-sdk/pkg/sdk/dev/install/lib/pulpissimo/pulpissimo -lrt -lrtio -lrt -lgcc
[... cut ...]
Debug: 279 125 command.c:651 run_command(): Command 'halt' failed with error code -4
User : 280 125 command.c:715 command_run_line(): 
Debug: 281 125 riscv.c:411 riscv_deinit_target(): riscv_deinit_target()
Error: 282 125 riscv.c:376 get_target_type(): Unsupported DTM version: 12
Debug: 283 125 target.c:1959 target_free_all_working_areas_restore(): freeing all working areas
Debug: 284 125 server.c:611 sig_handler(): Ignored extra Signal 13
Error: 285 125 remote_bitbang.c:106 remote_bitbang_quit(): fflush: Broken pipe
Error: 286 125 core.c:1487 adapter_quit(): failed: -4
```

bluewww, 2019-07-17:
Thanks I'm trying to fix this on the `fix_dm_rtl` branch, I reverted the offending commit. Also I'm running simulations testing `pulpissimo_compliance_test.cfg` and `pulpissimo_debug.cfg` to figure out if and when the tb gets stuck

irays18, 2020-12-30:
Hi is there any way to fix this issue other than switching branch to `fix_dm_rtl` ? Thank you

shivampotdar, 2021-03-01:
@bluewww can you please point to the `fix_dm_rtl` branches or give some idea for solving this issue. Thank you!

shivampotdar, 2021-03-01:
I just realised there are changes to `SimJTAG.sv` in the current master branch which seems to solve the above issue. `openocd` proceeds from there but I ran it for an hour and still didn't get a prompt for gdb. 
Am I missing something here? Is "a while" supposed to be more than hour in https://github.com/pulp-platform/pulpissimo/blob/master/rtl/tb/README.md#jtag-taps-and-flash

Thanks!

bluewww, 2021-03-01:
It is indeed extremly slow since openocd scans through 992 harts before finding the correct (working) one.

shivampotdar, 2021-03-02:
Got it. It does eventually reach to 992 but with GDB I am not able to connect - 

From GDB - 
```
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
```

From OpenOCD log -
```
Info : 13320 64559856 server.c:100 add_connection(): accepting 'gdb' connection on tcp/3333
Debug: 13321 64559856 breakpoints.c:358 breakpoint_clear_target_internal(): Delete all breakpoints for target: riscv.cpu
Debug: 13322 64559856 breakpoints.c:543 watchpoint_clear_target(): Delete all watchpoints for target: riscv.cpu
Debug: 13323 64559856 target.c:1609 target_call_event_callbacks(): target event 19 (gdb-attach) for core 992
Debug: 13324 64559856 target.c:4569 target_handle_event(): target(0): riscv.cpu (riscv) event: 19 (gdb-attach) action: halt
Debug: 13325 64559856 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_halt
Debug: 13326 64559856 command.c:143 script_debug(): command - halt ocd_halt
Debug: 13327 64559856 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (306022). This may cause trouble with GDB connections.
Debug: 13329 64559856 riscv.c:1550 riscv_openocd_poll(): polling all harts
Debug: 13330 64559856 riscv.c:2385 riscv_set_current_hartid(): setting hartid to 992, was 992
Debug: 13331 64559856 riscv.c:1504 riscv_poll_hart(): polling hart 992, target->state=2
Debug: 13332 64600784 riscv-013.c:393 scan(): 41b 3i r 00000000 @11 -> + 00000000 @11
Debug: 13333 64634049 riscv-013.c:393 scan(): 41b 3i - 00000000 @11 -> + 00030382 @11
Debug: 13334 64634049 riscv-013.c:404 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
Debug: 13336 64634049 target.c:3059 handle_halt_command(): -
Debug: 13337 64634049 riscv.c:1030 riscv_halt(): [992] halting all harts
Debug: 13338 64634049 riscv.c:963 halt_prep(): prep hart 992
Debug: 13339 64634050 riscv.c:2385 riscv_set_current_hartid(): setting hartid to 992, was 992
Debug: 13340 64675079 riscv-013.c:393 scan(): 41b 3i r 00000000 @11 -> + 00000000 @11
Debug: 13341 64709032 riscv-013.c:393 scan(): 41b 3i - 00000000 @11 -> + 00030382 @11
Debug: 13342 64709032 riscv-013.c:404 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
Debug: 13343 64709032 riscv.c:967 halt_prep(): Hart 992 is already halted.
Debug: 13344 64709032 riscv.c:2385 riscv_set_current_hartid(): setting hartid to 992, was 992
Debug: 13345 64751258 riscv-013.c:393 scan(): 41b 3i r 00000000 @11 -> + 00000000 @11
Debug: 13346 64786276 riscv-013.c:393 scan(): 41b 3i - 00000000 @11 -> + 00030382 @11
Debug: 13347 64786276 riscv-013.c:404 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
Debug: 13348 64786276 riscv.c:987 riscv_halt_go_all_harts(): Hart 992 is already halted.
Debug: 13349 64786276 riscv.c:2401 riscv_invalidate_register_cache(): [992]
Debug: 13350 64786276 target.c:1609 target_call_event_callbacks(): target event 0 (gdb-halt) for core 992
Debug: 13351 64786276 target.c:1609 target_call_event_callbacks(): target event 1 (halted) for core 992
Debug: 13352 64786276 riscv.c:1550 riscv_openocd_poll(): polling all harts
Debug: 13353 64786276 riscv.c:2385 riscv_set_current_hartid(): setting hartid to 992, was 992
Debug: 13354 64786276 riscv.c:1504 riscv_poll_hart(): polling hart 992, target->state=2
Debug: 13355 64830336 riscv-013.c:393 scan(): 41b 3i r 00000000 @11 -> + 00000000 @11
Debug: 13356 64864447 riscv-013.c:393 scan(): 41b 3i - 00000000 @11 -> + 00030382 @11
Debug: 13357 64864447 riscv-013.c:404 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
Debug: 13358 64864447 gdb_server.c:1014 gdb_new_connection(): New GDB Connection: 1, Target riscv.cpu, state: halted
Debug: 13359 64864447 gdb_server.c:1046 gdb_connection_closed(): GDB Close, Target: riscv.cpu, state: halted, gdb_actual_connections=0
Debug: 13360 64864447 target.c:1609 target_call_event_callbacks(): target event 6 (gdb-end) for core 992
Debug: 13361 64864447 target.c:1609 target_call_event_callbacks(): target event 20 (gdb-detach) for core 992
Info : 13362 64864447 server.c:577 server_loop(): dropped 'gdb' connection
Debug: 13363 64864447 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (304591). This may cause trouble with GDB connections.
Debug: 13365 64864447 riscv.c:1550 riscv_openocd_poll(): polling all harts
Debug: 13366 64864447 riscv.c:2385 riscv_set_current_hartid(): setting hartid to 992, was 992
```

shivampotdar, 2021-03-02:
I found this https://github.com/pulp-platform/pulpino/issues/140 and with this version of gdb, the log says :
```
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Bogus trace status reply from target: timeout
```