pulp-platform/pulpissimo #61: simulation tool porting recommendations

jeras, 2019-07-16:
Is there a recommended way to port the simulation to another tool? I have access to Cadence Incisive (maybe Xcelium) and Active-HDL (probably without support for SV classes).
Should I create a new make file (`make.incisive`?), how should I handle build file generators, ... what should generated file/folder names be?

Is the effort even worth it, would I hit too many roadblocks to bother? I know error/warning feedback is useful to the project, but I would like to get to a working simulation environment.

bluewww, 2019-07-19:
I've got an initial port working for xcelium. I'll start pushing it soon. There is obviousely not direct sdk support though so you have to play around with tcl scripts (for now?).

bluewww, 2019-07-23:
Preliminary support is merged in #66. Let me know if something is amiss.

closed 2019-07-23 by bluewww

FrischAd, 2019-08-09:
Thanks for the support of xcelium.
I tried it and the results are correct but I'm getting some errors executing the :

`
[TB] 11287801ns - Waiting for end of computation
ERROR:          SYSTF       BADMCD
         The MCD or file descriptor passed to $fflush is invalid.
        ../rtl/tb/tb_fs_handler.sv, 328: $fflush(IOFILE[core_index])
[STDOUT-CL31_PE0] Hello !
`

No start and end address:
`
[CORE] Core settings: PULP_SECURE =           1, N_PMP_ENTRIES =          16, N_PMP_CFG           4
START_ADDR[0] = 0xxxxxxxxx; END_ADDR[0] = 0Xxxxxxxxx
START_ADDR[1] = 0xxxxxxxxx; END_ADDR[1] = 0Xxxxxxxxx
[TB]       0ns - Asserting hard reset
`

I wondered if you have them, too?
I attached the [stim.txt](https://github.com/pulp-platform/pulpissimo/files/3485585/stim.txt)

PULPissimo: f33ecbcf17b2c7fdf3796d352ba3a8689603243f (Current version)
Xcelium: 19.03-s010
PULP-Builder: f9f262b374d0f44947f048d7651fa17b8de15aab (Current version)