pulp-platform/pulpissimo #62: Testbench in 32 bit cannot run on questasim 64 bit, issues with 64 bit toolchain build

vikramjain236, 2019-07-19:
I followed the readme file of pulpissimo and compiled the toolchain with the following command:
./configure --prefix=/opt/riscv --with-arch=rv32imc --with-cmodel=medlow --enable-multilib
make
When I run the tests (helloworld) I get the following errorLaunching simulator with command:

export VSIM_RUNNER_FLAGS="-gLOAD_L2=JTAG -permit_unmatched_virtual_intf -gBAUDRATE=115200" && vsim -64 -c -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl; set_tcheck; run_and_exit;'
Reading pref.tcl

# 10.7c

# source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do
# couldn't read file "/volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do": no such file or directory
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl
# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/volume1/users/vjain/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf "+VSIM_PATH=/volume1/users/vjain/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gBAUDRATE=115200 -gENABLE_DEV_DPI=0 -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 
# Start time: 12:16:11 on Jul 19,2019
# ** Error: (vsim-3816) Optimized design "vopt_tb" cannot be used by this version of the simulator - it was only compiled for a 32-bit version of vsim
# Error loading design
Error loading design
# End time: 12:16:11 on Jul 19,2019, Elapsed time: 0:00:00
# Errors: 1, Warnings: 0
Simulator reported an error, leaving
make: *** [run] Error 1

To solve this issue, I tried building the linux toolchain in 64 bit also using 
./configure --prefix=/opt/riscv --enable-multilib
make linux

But, I get the following errors

/tmp/ccQYOGlb.s: Assembler messages:
/tmp/ccQYOGlb.s:4963: Error: unrecognized opcode `lwu a1,.LANCHOR0+2'
/tmp/ccQYOGlb.s:5179: Error: unrecognized opcode `lwu a5,.LANCHOR0'
/tmp/ccQYOGlb.s:5257: Error: unrecognized opcode `lwu a4,.LANCHOR0+2'
/tmp/ccQYOGlb.s:5904: Error: unrecognized opcode `lwu a2,.LANCHOR0+2'
/tmp/ccQYOGlb.s:5996: Error: unrecognized opcode `lwu a3,.LANCHOR0+2'
make[4]: *** [unwind-dw2.o] Error 1
make[4]: Leaving directory `/volume1/users/vjain/pulp-riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/lib32/ilp32/libgcc'
make[3]: *** [multi-do] Error 1
make[3]: Leaving directory `/volume1/users/vjain/pulp-riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc'
make[2]: *** [all-multi] Error 2
make[2]: Leaving directory `/volume1/users/vjain/pulp-riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc'
make[1]: *** [all-target-libgcc] Error 2
make[1]: Leaving directory `/volume1/users/vjain/pulp-riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [stamps/build-gcc-linux-stage1] Error 2

bluewww, 2019-07-19:
It's a `vsim` issue, it doesn't have anything to do with `gcc`, so you don't need to touch it.
What is your `vsim` version? You probably need to specify a `-64` flag somewhere in the build process, which is done implicitely for newer questasim versions (which we probably assume).

vikramjain236, 2019-07-19:
Questasim 10.7c

bluewww, 2019-07-19:
Try to rebuild PULPissimo by running `make clean build` in the root of the project. Do you still get an error?

vikramjain236, 2019-07-19:
Yes, the same error.

vikramjain236, 2019-07-19:
Attached is the logs from the build on pulpissimo
[log.txt](https://github.com/pulp-platform/pulpissimo/files/3410826/log.txt)

bluewww, 2019-07-21:
So I tried to reproduce your issue but I can't. You seem to use a little bit newer version than me:
```
$ vsim -version
Questa Sim-64 vsim 10.7b_1 Simulator 2018.07 Jul 26 2018
```
From what I can infer it seems like `vsim` tries to run in 64-bit mode but `vlog` and `vopt` were run in 32-bit mode. So It might be that your newer version of `vsim` defaults to 64-bit mode. This can be fixed by putting a `-64` at the right place (the command which is used to launch should be modified to include this).

I think it is best if we put a `-64` everywhere anyway and maybe thats going to fix your issue too.

vikramjain236, 2019-07-24:
I got it working now by adding -64 option to vopt command in rtl_vopt.tcl file.

closed 2019-07-24 by vikramjain236

bluewww, 2019-07-24:
Ok good I will add this if it foesnt break anything for us

vikramjain236, 2019-07-24:
I ran into new problems while running the helloworld test. Can you have a look?. Here are the logs:

make clean all run
rm -rf /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c test.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test`
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o -nostartfiles -nostdlib -Wl,--gc-sections -L/volume1/users/vjain/pulp-builder/install/rules -Tpulpissimo/link.ld -L/volume1/users/vjain/pulp-builder/install/lib/pulpissimo -lrt -lrtio -lrt -lgcc 
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=**/rt/type=pulp-rt  --config-opt=**/rt/type=pulp-rt --dir=/volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo --binary=test/test prepare
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=**/rt/type=pulp-rt  --config-opt=**/rt/type=pulp-rt --dir=/volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo --binary=test/test
Launching simulator with command:
export VSIM_RUNNER_FLAGS="-gLOAD_L2=JTAG -permit_unmatched_virtual_intf -gBAUDRATE=115200" && vsim -64 -c -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl; set_tcheck; run_and_exit;'
Reading pref.tcl

# 10.7c

# source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do
# couldn't read file "/volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do": no such file or directory
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl
# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/volume1/users/vjain/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf "+VSIM_PATH=/volume1/users/vjain/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gBAUDRATE=115200 -gENABLE_DEV_DPI=0 -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 
# Start time: 14:28:48 on Jul 24,2019
# //  Questa Sim-64
# //  Version 10.7c linux_x86_64 Aug 17 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
[... cut ...]
# [TB] 11225701ns - Waiting for end of computation
# [STDOUT-CL31_PE0] Hello !
# [TB] 11463901ns - Received status core: 0x00000000
# ** Note: $stop    : /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_pulp.sv(782)
#    Time: 11463901 ns  Iteration: 0  Instance: /tb_pulp
# Break at /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 782
# Stopped at /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 782
# End time: 14:29:24 on Jul 24,2019, Elapsed time: 0:00:36
# Errors: 5, Warnings: 27

vikramjain236, 2019-07-24:
Hi bluewww, I ran into some new problems while running the helloworld test. Can you have a look? Here are the logs:

make clean all run
rm -rf /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c test.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test`
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o -nostartfiles -nostdlib -Wl,--gc-sections -L/volume1/users/vjain/pulp-builder/install/rules -Tpulpissimo/link.ld -L/volume1/users/vjain/pulp-builder/install/lib/pulpissimo -lrt -lrtio -lrt -lgcc 
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=**/rt/type=pulp-rt  --config-opt=**/rt/type=pulp-rt --dir=/volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo --binary=test/test prepare
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=**/rt/type=pulp-rt  --config-opt=**/rt/type=pulp-rt --dir=/volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo --binary=test/test
Launching simulator with command:
export VSIM_RUNNER_FLAGS="-gLOAD_L2=JTAG -permit_unmatched_virtual_intf -gBAUDRATE=115200" && vsim -64 -c -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl; set_tcheck; run_and_exit;'
Reading pref.tcl

# 10.7c

# source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do
# couldn't read file "/volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do": no such file or directory
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl
# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/volume1/users/vjain/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf "+VSIM_PATH=/volume1/users/vjain/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gBAUDRATE=115200 -gENABLE_DEV_DPI=0 -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 
# Start time: 14:28:48 on Jul 24,2019
# //  Questa Sim-64
# //  Version 10.7c linux_x86_64 Aug 17 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
[... cut ...]
# [TB] 11225701ns - Waiting for end of computation
# [STDOUT-CL31_PE0] Hello !
# [TB] 11463901ns - Received status core: 0x00000000
# ** Note: $stop    : /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_pulp.sv(782)
#    Time: 11463901 ns  Iteration: 0  Instance: /tb_pulp
# Break at /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 782
# Stopped at /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_pulp.sv line 782
# End time: 14:29:24 on Jul 24,2019, Elapsed time: 0:00:36
# Errors: 5, Warnings: 27

bluewww, 2019-07-24:
you mean this?
```
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
```
I have the same thing popping up, but its a `vsim` bug (as it says there) and its not telling me what the exact issue is. The only way to figure out the exact cause is to start removing code until you can narrow down the issue to the exact statement(s) which are responsible.

I can try older `vsim` versions to check if it's a regression of `vsim`.

vikramjain236, 2019-07-24:
Ok. I will have a look.

leduchuybk, 2021-08-11:
I have same problem when executing $make clean run all in hello directory.
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989).
Did you guys fix the problems?

kangliyu1, 2021-10-21:
> you mean this?
> 
> ```
> # ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
> # Please contact Questa support at http://supportnet.mentor.com
> # ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
> # Please contact Questa support at http://supportnet.mentor.com
> # ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
> # Please contact Questa support at http://supportnet.mentor.com
> # ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
> # Please contact Questa support at http://supportnet.mentor.com
> ```
> 
> I have the same thing popping up, but its a `vsim` bug (as it says there) and its not telling me what the exact issue is. The only way to figure out the exact cause is to start removing code until you can narrow down the issue to the exact statement(s) which are responsible. _Any help would really be appreciated!_
> 
> I can try older `vsim` versions to check if it's a regression of `vsim`. I don't get an error running `PULPissimo` on other simulators.
> 
> Anyway the simulation seems to return the correct result that's why I didn't bother looking closer into it.

Hello! Could you please change the questasim version to solve the above error?