pulp-platform/pulpissimo #69: Simulation of hwme accelerator stuck

vikramjain236, 2019-08-07:
I tried to run the hwme accelerator example in pulp-rt-examples. But, the simulation seems to get stuck at some point. I have successfully ran the helloworld example before. I use Questasim 10.7c.

/volume1/users/vjain/pulp-rt-examples/accelerators/hwme $ make clean all run
rm -rf /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/hwme /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc/hwme.o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc//volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc//volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_pad_conf.o  
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O2 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/fc_config.h    -MMD -MP -c hwme.c -o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc/hwme.o
hwme.c: In function 'main':
hwme.c:40:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
   uint32_t *a = (uint8_t *) 0x1c010000;
                 ^
hwme.c:41:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
   uint32_t *b = (uint8_t *) 0x1c010200;
                 ^
hwme.c:42:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
   uint32_t *c = (uint8_t *) 0x1c010400;
                 ^
hwme.c:43:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
   uint32_t *d = (uint8_t *) 0x1c010600;
                 ^
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O2 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_conf.c -o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc//volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_conf.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O2 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_pad_conf.c -o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc//volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/hwme`
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/hwme /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc/hwme.o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc//volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/hwme/fc//volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo/rt_pad_conf.o -nostartfiles -nostdlib -Wl,--gc-sections -L/volume1/users/vjain/pulp-builder/install/rules -Tpulpissimo/link.ld -L/volume1/users/vjain/pulp-builder/install/lib/pulpissimo -lrt -lrtio -lrt -lgcc -lbench 
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=rt/start-all=true  --config-opt=**/rt/fc-start=true  --config-opt=**/rt/type=pulp-rt  --config-opt=rt/start-all=true  --config-opt=**/rt/fc-start=true  --config-opt=**/rt/type=pulp-rt --dir=/volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo --binary=hwme/hwme prepare
pulp-run --config-file=pulpissimo@config_file=chips/pulpissimo/pulpissimo.json  --config-opt=platform=rtl  --config-opt=rt/start-all=true  --config-opt=**/rt/fc-start=true  --config-opt=**/rt/type=pulp-rt  --config-opt=rt/start-all=true  --config-opt=**/rt/fc-start=true  --config-opt=**/rt/type=pulp-rt --dir=/volume1/users/vjain/pulp-rt-examples/accelerators/hwme/build/hwme.c/pulpissimo --binary=hwme/hwme
Launching simulator with command:
export VSIM_RUNNER_FLAGS="-gLOAD_L2=JTAG -permit_unmatched_virtual_intf -gBAUDRATE=115200" && vsim -64 -c -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do' -do 'source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl; set_tcheck; run_and_exit;'
Reading pref.tcl

# 10.7c

# source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do
# couldn't read file "/volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do": no such file or directory
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl
# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/volume1/users/vjain/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf "+VSIM_PATH=/volume1/users/vjain/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gBAUDRATE=115200 -gENABLE_DEV_DPI=0 -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 
# Start time: 16:24:25 on Aug 07,2019
# //  Questa Sim-64
# //  Version 10.7c linux_x86_64 Aug 17 2018
# //
# //  Copyright 1991-2018 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Error: (vsim-191) Questa has encountered an unexpected internal error: ../../src/vsim/vsimfunc.c(1989). 
# Please contact Questa support at http://supportnet.mentor.com
# ** Warning: (vsim-3015) [PCDPC] - Port size (32) does not match connection size (64) for port 'axi_slave_w_data'. The port definition is at: /volume1/users/vjain/pulpissimo/sim/../ips/axi/axi_slice_dc/axi_slice_dc_slave.sv(56).
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/dc_fifo_datain_bus_i/axi_slice_i File: /volume1/users/vjain/pulpissimo/sim/../ips/axi/axi_slice_dc/axi_slice_dc_slave_wrap.sv Line: 50
[... cut ...]
# @1 [   x,    x,    x,    x,    x,    2] UPDATE CURRENT LOOP                      
# @2 [   x,    x,    x,    x,    x,    2] UPDATE CURRENT LOOP                      
# @3 [   x,    x,    x,    x,    x,    2] ITERATE CURRENT LOOP                     
# @0 [   x,    x,    x,    x,    x,    3] UPDATE CURRENT LOOP                      
# @1 [   x,    x,    x,    x,    x,    3] UPDATE CURRENT LOOP                      
# @2 [   x,    x,    x,    x,    x,    3] UPDATE CURRENT LOOP                      
# @3 [   x,    x,    x,    x,    x,    3] ITERATE CURRENT LOOP

haaani, 2019-08-07:
You are lucky I fount it yesterday :) The HWPE event ID changed from 46 to 140 in the hardware, the sdk has not been updated yet putting this line at the beginning of your program should solve the issue:

#define ARCHI_SOC_EVENT_FCHWPE0 140

vikramjain236, 2019-08-07:
Thanks! Works now.

closed 2019-08-07 by vikramjain236