pulp-platform/pulpissimo #71: Pulpissimo `make clean build` fails with QuestaSim

dynex817, 2019-08-14:
Hi all,

I built the toolchain using
`
./configure --prefix=/scratch.local/co96niro/pulpissimo/prefix --with-arch=rv32imc --with-cmodel=medlow --enable-multilib
`

So, when I now try to build the Pulpissimo repo with _make clean build_ as stated in the read me, I get the following error:

```
[...]
QuestaSim-64 vmap 10.6_1 Lib Mapping Utility 2017.01 Jan 30 2017
vmap vip_lib /storage/userA/pulpissimo/pulpissimo/install/modelsim_libs/vip_lib
Modifying modelsim.ini
QuestaSim-64 vmap 10.6_1 Lib Mapping Utility 2017.01 Jan 30 2017
vmap pulpissimo_lib /storage/userA/pulpissimo/pulpissimo/install/modelsim_libs/pulpissimo_lib
Modifying modelsim.ini
  Building L2_tcdm_hybrid_interco/soc_interconnect
  Building adv_dbg_if/adv_dbg_if
  Building apb2per/apb2per
  Building apb_adv_timer/apb_adv_timer
  Building apb_fll_if/apb_fll_if
  Building apb_gpio/apb_gpio
  Building apb_node/apb_node
  Building apb_interrupt_cntrl/apb_interrupt_cntrl
  Building axi_node/axi_node
  Building axi_slice/axi_slice
  Building axi_slice_dc/axi_slice_dc
  Building axi_mem_if/axi_mem_if
  Building timer_unit/timer_unit
  Building common_cells/common_cells_all
** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(578). Please contact Questa support at http://supportnet.mentor.com/
** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(580). Please contact Questa support at http://supportnet.mentor.com/
** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(580). Please contact Questa support at http://supportnet.mentor.com/
** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(580). Please contact Questa support at http://supportnet.mentor.com/
** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(580). Please contact Questa support at http://supportnet.mentor.com/
make[3]: *** [/storage/userA/pulpissimo/pulpissimo/install/modelsim_libs/common_cells_lib/common_cells_all.vmake] Error 2
make[2]: *** [build] Error 2
make[1]: *** [build] Error 2
make[1]: Leaving directory `/storage/userA/pulpissimo/pulpissimo/sim'
make: *** [build] Error 2
```

My vsim Version is "Questa Sim-64 vsim 10.6_1 Simulator"

What am I doing wrong? Or might it be the Questa Sim version?

Thanks in advance.

dynex817, 2019-08-19:
Now we got what was causing Questa Sim 10.6_1 to crash with id_queue.sv.

With the changes of
```
linked_data_d[head_tail_q[match_idx].head][0] = 1'b1;
```
to
```
linked_data_d[head_tail_q[match_idx].head].free = 1'b1;
```

and

```
linked_data_q[i][0] <= 1'b1;
```
to
```
linked_data_q[i].free <= 1'b1;
```

Questa Sim compiles the queue just fine.

So with the given minimal example
```
module foo
  #(
    parameter type data_t   = logic
    )

   (
    );

   typedef struct packed {
      data_t evil;
   } evil_struct;

   evil_struct x;

   always_comb begin
      // Access per Index crashes QuestaSim 10.6_1
      x[0] = 1'b1;
   end
endmodule // foo
```
it becomes clear that accessing an instantiated struct by index crashes Questa Sim while access by name does not. This only happens if the accessed element is defined as a parameter.

But with the simulation, the result is the same as before.

bluewww, 2019-08-19:
Thanks for the find! I will coordinate with the guys maintaining common_cells to get a fix out (that hopefully doesnt break other tools).

Atm I cant reproduce the exit_status error.

bennett78, 2020-04-23:
The above change works with 
$ vlog id_queue.sv 
Model Technology ModelSim - Intel FPGA Edition vlog 2020.1 Compiler 2020.02 Feb 28 2020
here is a diff
diff --git a/src/id_queue.sv b/src/id_queue.sv
index c432af4..61fdb31 100644
--- a/src/id_queue.sv
+++ b/src/id_queue.sv
@@ -197,7 +197,7 @@ module id_queue #(
                 if (oup_pop_i) begin
                     // Set free bit of linked data entry, all other bits are don't care.
                     linked_data_d[head_tail_q[match_idx].head]      = 'x;
-                    linked_data_d[head_tail_q[match_idx].head][0]   = 1'b1;
+                    linked_data_d[head_tail_q[match_idx].head].free = 1'b1;
                     if (head_tail_q[match_idx].head == head_tail_q[match_idx].tail) begin
                         head_tail_d[match_idx] = '{free: 1'b1, default: 'x};
                     end else begin
@@ -245,7 +245,7 @@ module id_queue #(
                 head_tail_q[i]      <= '{free: 1'b1, default: 'x};
                 // Set free bit of linked data entries, all other bits are don't care.
                 linked_data_q[i]    <= 'x;
-                linked_data_q[i][0] <= 1'b1;
+                linked_data_q[i].free <= 1'b1;
             end else begin
                 head_tail_q[i]      <= head_tail_d[i];
                 linked_data_q[i]    <= linked_data_d[i];
but 2020.1 doesn't have an "opt" function for
$ make opt
/s/opt/riscv/verilog/pulp-platform/pulpissimo/sim//tcl_files/rtl_vopt.tcl
Error: cannot find "/s/opt/Altera/intelFPGA_pro/modelsim_ase/bin/../linux/vopt"
child process exited abnormally
    while executing
"exec >@stdout vopt +acc=mnpr -o vopt_tb tb_pulp -floatparameters+tb_pulp -L /s/opt/riscv/verilog/pulp-platform/pulpissimo/sim/modelsim_libs/L2_tcdm_hy..."
    ("eval" body line 1)
    invoked from within
"eval exec >@stdout vopt +acc=mnpr -o vopt_tb tb_pulp -floatparameters+tb_pulp  $VSIM_IP_LIBS $VSIM_RTL_LIBS -work work "
    invoked from within
"if {[info exists ::env(VSIM_PATH)]} {
    #eval exec >@stdout vopt +acc=mnpr -o vopt_tb tb_pulp -floatparameters+tb_pulp -Ldir $::env(VSIM_PATH)/model..."
    (file "/s/opt/riscv/verilog/pulp-platform/pulpissimo/sim//tcl_files/rtl_vopt.tcl" line 26)
Makefile:37: recipe for target 'opt' failed
make: *** [opt] Error 1

bluewww, 2020-04-23:
There should be a `vopt` binary somewhere, unless intel removed it. This is what I get `QuestaSim-64 vopt 2020.1 Compiler 2020.01 Jan 28 2020`.

Also I can't reproduce the error, `vlog id_queue.v` doesn't complain at all with `10.6`, `10.7b` or `2020.1`

bennett78, 2020-04-24:
On 4/23/20 10:51 AM, bluew wrote:
    
    
      
      There should be a vopt binary somewhere, unless
        intel removed it. This is what I get QuestaSim-64 vopt
          2020.1 Compiler 2020.01 Jan 28 2020.
    
    Nope. vlog but no vopt....modelsim...not in intel distribution 20.1,
    19.4...I will attempt to contact them. Is there a free version of
    Mentor QuestaSim ?
    
    fwb@fw:/s/tmp$ which vopt
    /s/opt/Altera/intelFPGA_pro/modelsim_ase/bin/vopt
    fwb@fw:/s/tmp$ vopt -h
    Error: cannot find
    "/s/opt/Altera/intelFPGA_pro/modelsim_ase/bin/../linux/vopt"
    
    fwb@fw:/s/tmp$ which vlog
    /s/opt/Altera/intelFPGA_pro/modelsim_ase/bin/vlog
    fwb@fw:/s/tmp$ vlog -h
    Model Technology ModelSim - Intel FPGA Edition vlog 2020.1 Compiler
    2020.02 Feb 28 2020
    .....
    fwb@fw:/s/opt/Altera/intelFPGA_pro/modelsim_ase$ ll `find . -name
    vopt`
    lrwxrwxrwx 1 fwb fwb 6 Apr 23 12:39 ./bin/vopt -> ../vco
    fwb@fw:/s/opt/Altera/intelFPGA_pro/modelsim_ase$ ll `find . -name
    vlog`
    -rwxr-xr-x 1 fwb fwb 21745016 Feb 28 12:17 ./linuxaloem/vlog
    lrwxrwxrwx 1 fwb fwb        6 Apr 23 12:39 ./bin/vlog -> ../vco
    
    fwb@fw:/s/tmp$ uname -a
    Linux fw 4.15.0-48-generic #51-Ubuntu SMP Wed Apr 3 08:28:49 UTC
    2019 x86_64 x86_64 x86_64 GNU/Linux
    
    
      Also I can't reproduce the error, vlog id_queue.v
        doesn't complain at all with 10.6, 10.7b
        or 2020.1
    
    which OS ?
    
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/pulp-platform/pulpissimo/issues/71#issuecomment-618513301",
"url": "https://github.com/pulp-platform/pulpissimo/issues/71#issuecomment-618513301",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

bluewww, 2020-04-24:
I'm using Centos7 gnu/linux. I suspect that intel modelsim is actually a fork of when questasim was still called modelsim and thus doesn't get all the SystemVerilog updates.

What is the output of `vsim -help | grep vopt`?
On ~10.7b~ it looks like this to me:
```bash
vsim -help | grep vopt
   -nocoverage             Passed to vopt to turn off code coverage compile options
   -novopt                 Force incremental mode (pre-6.0 behavior)
                           (vopt -pdu) with an associated default SDF
   -vopt                   Run vopt optimization before elaborating the simulator
   -voptargs="<arglist>"   Pass the specified arguments to vopt
   -vopt_verbose           Display verbose informational messages
                           fixed-size arrays marked for randomization by vlog/vopt.
                           variables marked for randomization by vlog/vopt.
                           applying +initreg/+initmem options to vlog/vopt
   -no_autoacc             Prevents vsim from automatically passing the +acc switch to vopt.
```

FrancescoConti, 2020-04-24:
Intel ModelSim is the same as Altera ModelSim (Altera was bought by Intel). It only supports a subset of functionality of the full version. I think vopt is removed as well as a lot of DPI related functionality.

I know some people have managed to make parts of PULPissimo to work with it (browse other issues and the forums at pulp-platform.org) but in general we do not support it.

bennett78, 2020-04-24:
> Intel ModelSim is the same as Altera ModelSim (Altera was bought by Intel). It only supports a subset of functionality of the full version. I think vopt is removed as well as a lot of DPI related [...]

Not sure why modelSim, quartus is referenced, only if the target is  an Altera FPGA.
I'm looking for a consistent process: RISC-V, FP32, (system)Verilog, minimum VHDL, XIlinx, Altera, Lattice FPGA,  TSMC ASIC, open source, etc...maybe the  README.md should mention a map, path, recommended tools to target(s) : HDL language, verification method, simulators, synthesis, target boards, git branches,  etc

bennett78, 2020-04-24:
> I'm using Centos7 gnu/linux. I suspect that intel modelsim is actually a fork of when questasim was still called modelsim and thus doesn't get all the SystemVerilog updates.

Centos is a build of a previous version Redhat. I prefer xUbuntu, LTS as more uptodate with new design efforts.
> I tried to download a free questasim -> http://allpcworld.com/download-mentor-graphics-xpedition-enterprise-vx-2-free/ -> http://alltutorialsworld.com/ ???

> What is the output of `vsim -help | grep vopt`?
> On `10.7b` it looks like this to me:
> 
> ```shell
> vsim -help | grep vopt
NULL
> -nocoverage             Passed to vopt to turn off code coverage compile options
>    -novopt                 Force incremental mode (pre-6.0 behavior)
>                            (vopt -pdu) with an [...]

bluewww, 2020-04-24:
I think the FPGA section is quite complete in what boards PULPissimo runs on.

@MarekPikula ported PULPissimo to run with Quartus, but in general we don't support it because the Intel tools do not support SystemVerilog 2008 properly despite claiming to do so.

As for what concerns simulators the big three should work just fine, but currently we ony have questa and xcelium support in master. A lot of the IPs also work with verilator, but PULPissimo itself doesn't. I think its feasible to port it.

We did tape out versions of PULPissimo and PULP (similar codebases) in tscm65, umcL65 and gf22 iirc.

bluewww, 2020-04-24:
> > I'm using Centos7 gnu/linux. I suspect that intel modelsim is actually a fork of when questasim was still called modelsim and thus doesn't get all the SystemVerilog updates.
> 
> Centos is a build of a previous version Redhat. I prefer xUbuntu, LTS as more uptodate with new design efforts.

Centos7 is officially supported by most tool vendors, which is why we run it. We do also support ubuntu16 to some degree.
> 
> > I tried to download a free questasim -> http://allpcworld.com/download-mentor-graphics-xpedition-enterprise-vx-2-free/ -> http://alltutorialsworld.com/ ???

FrancescoConti, 2020-04-24:
> Not sure why modelSim, quartus is referenced, only if the target is an Altera FPGA.

I think the README is already clear on the matter of ModelSim support:
https://github.com/pulp-platform/pulpissimo#requirements

> I'm looking for a consistent process: RISC-V, FP32, (system)Verilog, minimum VHDL, XIlinx, Altera, Lattice FPGA, TSMC ASIC, open source, etc...maybe the README.md should mention a map, path, [...]

I have no idea what you are talking about by putting together "RISC-V, FP32, (system)Verilog, minimum VHDL, XIlinx, Altera, Lattice FPGA, TSMC ASIC, open source, etc".

In terms of targets, to compound @bluewww , we do not know all PULP-based tapeouts, because anyone can take the code and use it. The PULP/PULPissimo-based chips we know about have also been using TSMC 55nm (https://en.wikichip.org/wiki/greenwaves/gap8), ST 28nm FD-SOI (http://asic.ee.ethz.ch/2015/Pulpv3.html), GF 40nm (http://asic.ee.ethz.ch/2015/Honey_Bunny.html), and TMSC 40nm (http://asic.ee.ethz.ch/2017/Mr.Wolf.html).
The code is synthesizable, so basically the only limit to this is your fantasy (and the depth of your pocket).

HarshitSharm-a, 2020-07-20:
Hi,
I am facing same issue as @dynex817.  I used @dynex817 solution and made the changes in id_queue.sv.

After that when I run the hello world example, I got an "Error: (vish-4014) No objects found matching 'sim:/tb_pulp/exit_status'."

[image]

The output of **vsim -help | grep vopt** is as follows:

> -nocoverage                Passed to vopt to turn off code coverage compile options
>  -novopt                        Force incremental mode (pre-6.0 behavior)
>                                      (vopt -pdu) with an associated default SDF
>  -vopt                            Run vopt optimization before elaborating the simulator
>  -voptargs="<arglist>"   Pass the specified arguments to vopt
>  -vopt_verbose             Display verbose informational messages
>                                     This option should be used with full design visibility e.g. vopt +acc
>                                     fixed-size arrays marked for randomization by vlog/vopt.
>                                     variables marked for randomization by vlog/vopt.
>                                     applying +initreg/+initmem options to vlog/vopt
>   -no_autoacc              Prevents vsim from automatically passing the +acc switch to vopt.

I am using Centos6. 

Thanks in advance.

closed 2020-11-17 by bluewww