pulp-platform/pulpissimo #74: jtag trst pin on fpga implementation

AB036, 2019-08-29:
In the ZCU104 pulpissimo instantiation, the jtag trst pin isn't connected to an external connector; it is set to a constant 1 instead.

File  pulpissimo/fpga/pulpissimo-zcu104/rtl/xilinx_pulpissimo.v :
`            .pad_jtag_trst(1'b1),
`

Is this normal?
What is the reason behind this?

Thank you

meggiman, 2019-09-06:
@marcowidmer can you comment on that?

marcowidmer, 2019-09-21:
I used the Digilent JTAG-HS1 cable which does not support trst, so I simply tied the signal to 1. If you want, you can route it to a GPIO, but you have to make sure that it is pulled up in case nothing is connected. Anyway, programming worked fine without this signal, I believe that there is a soft reset in the JTAG protocol that is used instead.

AB036, 2019-09-25:
I eventually used an other card with GPIOs and the OpenOCD sysfsgpio driver as my JTAG interface, and left the TRST pin tied to 1. This works fine functionally.
Thank you for the answer.

closed 2019-09-25 by AB036