pulp-platform/pulpissimo #75: get error when i "make clean Build"

OmarDez, 2019-09-04:
I get an error message when I "make clean build" in the building rtl platform stage, I have uploaded screenshot of this error.

Regards.
[image]

bluewww, 2019-09-04:
Did you run `./generate_scripts` before calling `make`?

OmarDez, 2019-09-04:
yes, I first run `./update-ips`, and after that  `./generate-scripts`, but it displays an error

OmarDez, 2019-09-04:
> yes, I first run `./update-ips`, and after that `./generate-scripts`, but it displays an error

Gives the same error

ranaya123, 2019-09-10:
Hi, I can reproduce the same error when building Pulpissimo platform with pulp-sdk (not pulp-builder). 
**ERROR: there are unallowed keys for ip 'fpu_div_sqrt_mvp', sub-ip 'div_sqrt_top_mvp':**

This is the ./update-ips output :
https://pastebin.com/KPEfTZHC

This is same for master branch as well as newer versions of Pulpissimo (i.e. v4). What could possibly go wrong here ?

bluewww, 2019-09-10:
These are recent patches that added more entries to the `src_files.yml` for other tools. I'm not sure if this is the cause of the breakage your experiencing.

OmarDez, 2019-09-10:
How can I install correctly the system?

bluewww, 2019-09-10:
It looks like your issue is related to this one. It is this PR https://github.com/pulp-platform/fpnew/pull/18 from yesterday.

bluewww, 2019-09-10:
Ok I can reproduce both errors, I will try to fix it asap (most likely tomorrow).

OmarDez, 2019-09-10:
Thanks very much, I´ll wait for the fix
Regards.

ranaya123, 2019-09-11:
@OmarDez : If you want a quick fix and get the rtl simulations up and running with Modelsim, change the "generate_script" file content to following (This was in-fact the old gen script) :
https://pastebin.com/0RWQdm2b

As @bluewww , stated the issue stems from the new stuffs/support added to the generate script.

bluewww, 2019-09-11:
Fixed in https://github.com/pulp-platform/IPApproX/commit/7c356b29d8f3a8c06b326dbadd269907503b9348.

OmarDez, 2019-09-11:
Sorry for the question, do I have to clone everything again and install everything in order for the changes to take place?

OmarDez, 2019-09-11:
@ranaya123 I change the "generate_Script" file but the error continues.
@bluewww Do I have to install everything from the begining ? I have updated the pulpissimo folder and followed the steps of `./update-ips`  and `./generate-scripts`, but it gives me the same error.
Thanks

bluewww, 2019-09-11:
Unfortunately ipstools is not very smart. Its best if you clone and run the scripts again. If you don't want to do that you can try to delete ipstools in the folder.

OmarDez, 2019-09-11:
The scripts run correctly and clone the file with the updated code, but the same error message when I run `make clean build` continues, is the same as the screen shot I uploaded before.

bluewww, 2019-09-11:
I can't reproduce your problem. I run these commands
```
git clone https://github.com/pulp-platform/pulpissimo/
cd pulpissimo
./update-ips
make clean build

```

OmarDez, 2019-09-11:
@bluewww I run the commands you write and gives me the error,.
I add you a screen shot when I run `./scripts/build-runtime` is this the expected result? 
[image]
After that I run these commands:
`source sdk-setup.sh`
`source configs/rtl.sh`
`cd ..`
`cd pulpissimo`
`./update-ips`
`make clean build`
and gives me error

FrancescoConti, 2019-09-11:
These keys have been added by a PULP partner for another project, but apparently they break compatibility for some people. We are currently working to fix this — in the mean time the workaround is to go inside the offending ips in `ips/ip_name` and remove all the section with `jg*` references.

bluewww, 2019-09-11:
@FrancescoConti this is already fixed, see my previous comments

OmarDez, 2019-09-11:
@bluewww  I also have this error while installing SDK
[image]
Thanks

OmarDez, 2019-09-12:
I tried to build the SDK with `make all` in two different computer and it showed the same error "Permission denied (publickey)" I tried to install PULP in another computer but I couldn't build the SDK, is there a solution for this??
Thanks. Regards.

OmarDez, 2019-09-12:
this is the screen shot [image]
Regards.

ranaya123, 2019-09-12:
@OmarDez : Those errors are related to your public ssh key of github. If you followed the steps for pulp-sdk build, then the build should work !
Thanks @bluewww for the quick fix !

bluewww, 2020-01-27:
Looks solved. Please reopen if needed.

closed 2020-01-27 by bluewww