pulp-platform/pulpissimo #78: Quartus support

MarekPikula, 2019-09-19:
I'm working on making the project's codebase compatible with Intel/Altera Quartus for synthesis. Although Intel claims support for SystemVerilog-2009 it's not entirely true (I'm currently documenting all the stuff I've stumbled upon [here](https://github.com/MarekPikula/quartus-sv-gotchas)).

I'm preparing patches, which make the code synthesizing correctly. So far without any automated scripts for project creation and bitstream generation.

I'd like to ask if the project is interested in such patches.

bluewww, 2019-09-19:
I just looked through your [document](https://github.com/MarekPikula/quartus-sv-gotchas/blob/master/Intel%20Quartus%20SystemVerilog%20gotchas.md) and the changes are unfortunately incompatible with the [styleguide ](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md) we use. Using sv-2009 instead of sv-2012 or sv-2017 is fine, but the stuff you point out in your document shows that quartus' sv support is horribly broken.

I think using `if else` instead of `unique case` is a no-go since they are not equivalent. Something with `case` plus `//parallel_case` would be needed. More information can be found [here](http://www.sunburst-design.com/papers/CummingsSNUG2005Israel_SystemVerilog_UniquePriority.pdf).

While I appreciate your effort its unlikely that we can accept your patches into `master`.

What might be possible is that you can put your work into a branch (your patched work with the helpful guiding documents) which we can refer to in the `README.md` in master.

MarekPikula, 2019-09-19:
Sure, to be honest that's the response I expected :wink: I'll prepare all the necessary patches and you'll see which changes can go to master and which not.

Which `unique case` issue are you referring to? (because there are multiple)

bluewww, 2019-09-19:
I was referring to the `unique case` example in your document

MarekPikula, 2019-09-19:
You mean "12.5.2 Constant expression in case statement"? There are also "case defaults sometimes not working" and "`unique case inside` not supported".

MarekPikula, 2019-09-19:
According to [this thread on SO](https://stackoverflow.com/questions/15418636/case-statement-in-verilog) it is equivalent as constant expression cases are to be evaluated in order of statements.

bluewww, 2019-09-19:
Indeed I'm referring to the example in `12.5.2 Constant expression in case statement`.

bluewww, 2019-09-19:
There is a `unique` infront of the `case`, meaning all the statements can/are evaluated in parallel.

MarekPikula, 2019-09-19:
In provided example it's clear that one of the statements must "win", since they write to the same wire. In this case this statement must have some sort of priority context. I will see how these two versions differ in terms of RTL in Vivado shortly.

MarekPikula, 2019-09-19:
To my surprise it's not evaluated in the same way, although resulting logic is pretty much the same (in this case). I agree though that in other cases it might not be true.

Here is with `case`:
[image]

Here is with `if ... else`:
[image]

MarekPikula, 2019-09-19:
Which branch should I base on for `apb_gpio`? `pulp_soc/ips_list.yml` point to detached commit.

bluewww, 2019-09-19:
Ugh, that's not good. Ideally I think we first point to pulpissimo to the `apb_gpio` (if still compatible)

MarekPikula, 2019-09-19:
What do you mean? In this specific context both results are equivalent taking into account logic that uses `exception_pc`.

I'm not sure I understand the second part of your comment. What compatibility are you referring to?

bluewww, 2019-09-19:
> To my surprise it's not evaluated in the same way, although resulting logic is pretty much the same (in this case). I agree though that in other cases it might not be true.
> 
> Here is with `case`:
> [...]
> 
> Here is with `if ... else`:
> [...]
> 
> FYI Quartus evaluates `if ... else` in the same way.

This looks fine to me I was not complaining about the results.
First and foremost both formulations are obviously logically equivalent, so simulation results are not impacted.

 What I was getting at is that with `unique case` you force tools to acknowledge that you want parallel logic or otherwise it has to warn you (e.g. when statements logically overlap during simulation). An `if else` or `case` statement implies a priority encoding, which _sometimes_ tools detect to be able to turn into parallel logic (for synthesis). If you are sure that the your statements can be parallel you can force some tool with `//parallel_case` to regards them as such.

MarekPikula, 2019-09-20:
Ah, I see your point. Later I'll experiment with different approaches and we'll see if I'll come with some universal solution.

MarekPikula, 2019-09-20:
I updated the description with list of cores requiring changes in my configuration. Are you @bluewww one of maintainers in all of them? I'll send patches in the next few weeks and then we'll decide which changes can go upstream and which should land in separate branch.

MarekPikula, 2019-09-20:
Could you @bluewww indicate against which branches should I prepare my patches?

MarekPikula, 2019-09-20:
That's all for today. Other changes are either significant or I don't know which branch should I base on.

MarekPikula, 2019-10-07:
Would you @bluewww consider bumping `tech_cells_generic` to master? I can see that master is much more polished than current version and for example doesn't introduce duplicate `pulp_clock_mux2`, what is a problem for Quartus.

MarekPikula, 2019-10-07:
And please point out what branch I should PR against for the rest of cores.

bluewww, 2019-10-08:
@MarekPikula Currently this PR is in progress for `pulp_soc` (and PULPissimo) that updates all IPs: https://github.com/pulp-platform/pulp_soc/pull/12. Currently still being tested.

You can see the cores/IPs that will be used soon so ideally PRs would target that version

MarekPikula, 2019-10-08:
What about [pulp-platform/apb_gpio](https://github.com/pulp-platform/apb_gpio)? Should I target `64bit` branch?

bluewww, 2019-10-08:
PULPissimo is still using a quite old version of `apb_gpio`. The update will either target `master` or `64bit` but I have to figure that out yet.

MarekPikula, 2019-10-08:
Actually PULPissimo uses pulp-platform/apb_gpio@730a9204dbb0d7057f45ef833d0a6e868c46107b, which is from 64bit branch.

[image]

bluewww, 2019-10-08:
Oh I see. I'm currently in touch with the guy who was doing development on it.

bluewww, 2019-10-10:
We will be targeting the `64bit` branch on `apb_gpio` but it will be merged to master anyway. I pinged the maintainers to cleanup the branch and merge it.

MarekPikula, 2019-10-14:
I figured that `axi_node` is not instantiated in PULPissimo, so I won't bother preparing patch, since I have no means and time resources to test it.

MarekPikula, 2019-10-15:
I have a question about `common_cells`. It's one of those breaking changes, so it will land in separate branch, since Quartus doesn't support [type parameters](https://github.com/MarekPikula/quartus-sv-gotchas/blob/master/Intel%20Quartus%20SystemVerilog%20gotchas.md#6203-type-parameters) used by various modules (including CDC used by `riscv-dbg`).

Since `common_cells` are used in various other projects, PULPissimo isn't using all modules present in repo. Should I make changes to all modules, or only to those used by PULPissimo?

bluewww, 2019-10-16:
I would start off with just the ones referred to by PULPissimo. I don't know how far you want to expand your quartus project.

MarekPikula, 2019-10-16:
My assignment was to evaluate just the PULPissimo on Cyclone V for comparison to previous core and I probably won't evaluate other projects from PULP Platform any time soon.

MarekPikula, 2019-10-16:
I guess with my commits and documentation, if anyone will want to port other projects to work with Quartus, will know what to do.

MarekPikula, 2019-10-17:
Could you create a `quartus` branch in `common_cells` based on `v1.13.1` flag? I'd like to push a PR.

bluewww, 2019-10-17:
Done https://github.com/pulp-platform/common_cells/tree/quartus

MarekPikula, 2023-06-03:
Hi @bluewww, sorry to dig up an old issue. I was cleaning up my GitHub PR list and noticed all those unmerged Quartus compatibility changes. Please let me know if they are even considered to be merged or if I should simply close all those PRs. I'm no longer interested in these changes, as I don't do any active development on Intel FPGAs.

bluewww, 2023-06-03:
Hi @MarekPikula . Unfortunately, we don't have the developer bandwidth (and actually also the hardware and tooling) to get your stuff merged. There have been some less intrusive changes which we merged but if there are no real users then we won't be able tell if we are breaking things for quartus anyway.

I suggested you close your PRs.

closed 2023-06-04 by MarekPikula