pulp-platform/pulpissimo #85: Duplicate definition of pulp_clock_gating_xilinx

MarekPikula, 2019-10-07:
It's defined both in `ips/tech_cells_generic/src/deprecated/pulp_clock_gating_async.sv` and `fpga/pulpissimo-*/rtl/pulp_clock_gating_xilinx.sv`. Fox Vivado it's no problem since it generates only a warning and in either case the definition is the same. For Quartus though it is a problem and synthesis cannot be completed.

Which one should be removed? I wonder if one from `tech_cells_generic`, since it's in `deprecated` folder anyways, or the one from `pulpissimo`, since it's just a one module in file?

MarekPikula, 2019-10-07:
Since it's problem with Quartus I'm marking #78 issue here for reference.

bluewww, 2019-10-08:
Is this fixed when we update `tech_cells_generic`?

MarekPikula, 2019-10-08:
Not really since `pulp_clock_gating` still exists in `src/pulp_clock_gating_xilinx.sv` in `tech_cells_generic`.

bluewww, 2019-11-13:
I think we remove the one from this repo. The deprecated folder in common_cells is to discourage people from using the cells inside in new projects.