pulp-platform/pulpissimo #87: Cleaner core configuration

MarekPikula, 2019-10-08:
What do you think about making more general approach to `CORE_TYPE`? Currently I'm working on evaluating [syntacore/scr1](https://github.com/syntacore/scr1) with PULPissimo and it looks a little bit messy to add another numbers for all different configurations of `scr1`.

I think the best approach would be to make `CORE_TYPE` for core alone as an enum, without features (so for example RI5CY, Ibex, scr1, …) and make additional enum or struct with additional features, like supported ISA and/or core-specific features.

If you consider it a good idea I could prepare patches.

bluewww, 2019-10-08:
Sounds like a good idea. Let me pull in @meggiman on this.

MarekPikula, 2019-10-21:
So what are your thoughts? I'm preparing patch for `scr1`, so I could start working on new configuration scheme shortly.

MarekPikula, 2019-10-25:
Here is a reference to my PR in scr1: syntacore/scr1#18

MarekPikula, 2019-11-07:
My PR has been merged. Any thoughts on this issue?

bluewww, 2019-11-13:
Looks good! Are there any difficulties with integrating this into PULPissimo? Also it looks line we need to fork the repo anyway to add the necessary files to make it work with our IPapprox flow or use patches.

MarekPikula, 2019-11-13:
Not really. It comes with integrated JTAG TAP, so I needed to replace current `riscv-dbg` and instantiated in FC subsystem.

I can work on integrating this, but first we'd need to have this cleaner core configuration for current cores.

From what they told if there are any additional PULPissimo-specific files they can be merged as well since it's either way in `pulpissimo_integration` branch.