pulp-platform/pulpissimo #88: L2 load from SPI flash

vikramjain236, 2019-10-09:
Hi,

I have been trying to run my accelerator on the pulpissimo environment, but I am facing some issues.
1. My code and stimuli arrays are very large. When I try to run the RTL simulation platform, it gives me this Illegal instruction issue. So, I changed the entry point of the code to address which I got from the elf file. I was able to get the thing working with a reduced data. After that, I tried to run in with a larger data in my code, changed the entry point and again I see the same issue about Illegal instructions. I guess, boot from JTAG does not really allow a very large binary.
2. So I decided to move to SPI flash and downloaded the model from Cypress, added it to the rtl/vip/ folder, uncommented the SPI flash part in the src_files.yml, changed USE_S25FS256S_MODEL to 1 and LOAD_L2 to STANDALONE and then ran the generate-scripts and make build.
3. Then I ran this command from the pulp-rt-examples/accelerators/hwme folder, which has my C code and stimuli header files

make conf boot=rom all run CONFIG_OPT=vsim/boot-mode=rom gui=1

But I get these prints:

# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-3040) Command line generic/parameter "SPI_FLASH_LOAD_MEM" not found in design.
# ** Warning: (vsim-3040) Command line generic/parameter "USE_SDVT_I2S" not found in design.
# ** Warning: (vsim-3040) Command line generic/parameter "USE_SDVT_SPI" not found in design.
#  set_tcheck
# invalid command name "set_tcheck"
#
run -all
# ** Warning: (vsim-7) Failed to open readmem file "s25fs256sOTP.mem" in read mode.
# No such file or directory. (errno = ENOENT)    : /volume1/users/vjain/pulpissimo/sim/../rtl/vip/spi_flash/S25fs256s/model/s25fs256s.v(998)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/genblk3/i_spi_flash_csn0
# ** Warning: (vsim-PLI-3407) Too many data words read on line 2049 of file "./boot/boot_code.cde". (Current address [2048], address range [0:2047])    : /volume1/users/vjain/pulpissimo/sim/../ips/tech_cells_generic/src/deprecated/generic_rom.sv(31)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/boot_rom_i/rom_mem_i
# [CORE] Core settings: PULP_SECURE =           1, N_PMP_ENTRIES =          16, N_PMP_CFG           4
# ** Warning: (vsim-8315) No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 0  Process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/lFC_CORE/RISCY_PMP/pmp_unit_i/<anonymous> File: /volume1/users/vjain/pulpissimo/sim/../ips/riscv/./rtl/riscv_pmp.sv Line: 677
# START_ADDR[0] = 0x1a100000; END_ADDR[0] = 0X1a400000
# START_ADDR[1] = 0x10000000; END_ADDR[1] = 0X10400000
# ** Warning: (vsim-8315) No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 0  Process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/jtag_lint_arbiter_i/<anonymous> File: /volume1/users/vjain/pulpissimo/sim/../ips/pulp_soc/rtl/components/tcdm_arbiter_2x1.sv Line: 49
# [TB]       0ns - Asserting hard reset
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_0_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_CLUSTER
# ** Warning: (vsim-3533) [FOFIW] - Failed to open file "fs/file_31_0.txt" for writing.
# No such file or directory. (errno = ENOENT)    : /volume1/users/vjain/pulpissimo/sim/../rtl/tb/tb_fs_handler.sv(269)
#    Time: 0 ps  Iteration: 0  Instance: /tb_pulp/i_fs_handler/FS_PRINTF_FC
# ** Warning: (vsim-8315) No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 1  Process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/lFC_CORE/RISCY_PMP/pmp_unit_i/<anonymous> File: /volume1/users/vjain/pulpissimo/sim/../ips/riscv/./rtl/riscv_pmp.sv Line: 677
# ** Warning: (vsim-8315) No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 1  Process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/jtag_lint_arbiter_i/<anonymous> File: /volume1/users/vjain/pulpissimo/sim/../ips/pulp_soc/rtl/components/tcdm_arbiter_2x1.sv Line: 49
# ** Warning: (vsim-8315) No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 2  Process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/fc_subsystem_i/FC_CORE/lFC_CORE/RISCY_PMP/pmp_unit_i/<anonymous> File: /volume1/users/vjain/pulpissimo/sim/../ips/riscv/./rtl/riscv_pmp.sv Line: 677
# ** Warning: (vsim-8315) No condition is true in the unique/priority if/case statement.
#    Time: 0 ps  Iteration: 12  Process: /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/jtag_lint_arbiter_i/<anonymous> File: /volume1/users/vjain/pulpissimo/sim/../ips/pulp_soc/rtl/components/tcdm_arbiter_2x1.sv Line: 49
# [TB]  514701ns retrying debug reg access
# [TB]  544001ns retrying debug reg access
# [TB]  573301ns retrying debug reg access
# [TB]  602601ns retrying debug reg access
# [TB]  631901ns retrying debug reg access
# [TB]  661201ns retrying debug reg access
# [TB]  690501ns retrying debug reg access
# [TB]  719801ns retrying debug reg access
# [TB]  749101ns retrying debug reg access

These retrying debug reg access prints doesn't stop. Looking at the code, these are from jtag_pkg.sv when doing read or write debug reg. Why does the code still uses JTAG interface even after I have changed the boot method to SPI flash? Does the current Pulpissimo environment have support for SPI flash?

quangdaovu, 2019-11-12:
Hi,

Have you managed to make it work yet? I'm working on the boot code for PULPissimo, but I also couldn't run the testbench with Spansion SPI flash & boot code from https://github.com/pulp-platform/boot-code.

On my testbench, `boot_addr` is `0x1a000080` but IF stage doesn't try to fetch instruction from boot_rom.

Regards,
Quang Dao

vikramjain236, 2019-11-12:
I did not try to run this again. I am working with JTAG for the time being.

vikramjain236, 2019-11-25:
Hi Quang,

I saw in the latest pulpissimo build, they have included the SPI flash in the generate scripts. In addition to instantiating the module in the pulpissimo environment, you need to run the generate scripts with the flash-vip option. I think that should make the thing work.
./generate-scripts --flash-vip

bluewww, 2019-11-26:
hello vikramjain236,
the scripts won't help you much since they rely on repositories on our internal gitlab instance. We have to do this for legal reasons since the VIP's are proprietary, though its probably not hard to edit the script to make it pull directly from the manufacturer's websites.

closed 2021-05-06 by bluewww