pulp-platform/pulpissimo #89: Error while building tests: cannot find -lrt

abdullahyildiz, 2019-10-15:
I installed the toolchain by following the instructions at [pulp-riscv-gnu-toolchain](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain#installation-pulp):

```
./configure --prefix=/opt/riscv --with-arch=rv32imc --with-cmodel=medlow --enable-multilib
make
```

When I try to build the _hello_ example at [pulp-rt-examples](https://github.com/pulp-platform/pulp-rt-examples), I get the following error:

```
cd pulp-rt-examples/hello
make clean all run
```

```
/home/user-1/Desktop/work/pulp/pulp-riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lrt
/home/user-1/Desktop/work/pulp/pulp-riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lrtio
/home/user-1/Desktop/work/pulp/pulp-riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lrt
collect2: error: ld returned 1 exit status
```

What is the correct way to install pulp-riscv-gnu-toolchain? I used the commit https://github.com/pulp-platform/pulp-riscv-gnu-toolchain/commit/34c464f8b60f7dd4fb495831c1221348d9890cd1 to install the toolchain.

MarekPikula, 2019-10-15:
You should build SDK as described in [README.md](https://github.com/pulp-platform/pulpissimo#prerequisites).

abdullahyildiz, 2019-10-15:
Is it sufficient to run these commands to build the pulp-riscv-gnu-toolchain:

```
./configure --prefix=/opt/riscv --with-arch=rv32imc --with-cmodel=medlow --enable-multilib
make
```

closed 2019-10-15 by abdullahyildiz