pulp-platform/pulpissimo #91: Regarding PULPissimo Installation

shillpa-j, 2019-10-17:
Hello..
I have been trying to install PULPissimo from past 2 months. Since I'm a starter I'm unable to rectify its errors. I have completed the installation process till GVSoC build as per the steps given in the Github. While executing the commands of runtime build which is the further process of installation I'm getting the following error. So, what may be the further steps to overcome this error
 command executed:./scripts/build-runtime
errors oocured:
scons: done building targets.
make: Leaving directory '/root/pulpissimo/pulp-builder/hal'
+ make -C pulp-rules all BUILD_DIR=/root/pulpissimo/pulp-builder/build/pulp-rules INSTALL_DIR=/root/pulpissimo/pulp-builder/install/ws TARGET_INSTALL_DIR=/root/pulpissimo/pulp-builder/install
make: Entering directory '/root/pulpissimo/pulp-builder/pulp-rules'
make: Nothing to be done for 'all'.
make: Leaving directory '/root/pulpissimo/pulp-builder/pulp-rules'
+ make -C pulp-rt header build install BUILD_DIR=/root/pulpissimo/pulp-builder/build/pulp-rt INSTALL_DIR=/root/pulpissimo/pulp-builder/install/ws TARGET_INSTALL_DIR=/root/pulpissimo/pulp-builder/install
make: Entering directory '/root/pulpissimo/pulp-builder/pulp-rt'
make: Nothing to be done for 'header'.
/bin/riscv32-unknown-elf-gcc  -march=rv32imcxgap8 -D__riscv__ -D__RT_USE_IO=1 -Os -g -fno-jump-tables -Werror -fno-tree-loop-distribute-patterns  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/root/pulpissimo/pulp-builder/install/include/io -I/root/pulpissimo/pulp-builder/install/include -include /root/pulpissimo/pulp-builder/build/pulp-rt/gap_rev1/cl_config.h    -DLANGUAGE_ASSEMBLY -MMD -MP -c kernel/riscv/pe-eu-v3.S -o /root/pulpissimo/pulp-builder/build/pulp-rt/gap_rev1/rt/cl/kernel/riscv/pe-eu-v3.o
make: /bin/riscv32-unknown-elf-gcc: Command not found
/root/pulpissimo/pulp-builder/build/pulp-rt/gap_rev1/__rules.mk:65: recipe for target '/root/pulpissimo/pulp-builder/build/pulp-rt/gap_rev1/rt/cl/kernel/riscv/pe-eu-v3.o' failed
make: *** [/root/pulpissimo/pulp-builder/build/pulp-rt/gap_rev1/rt/cl/kernel/riscv/pe-eu-v3.o] Error 127
make: Leaving directory '/root/pulpissimo/pulp-builder/pulp-rt'

 Thanks and Regards
Shilpa

haugoug, 2019-10-17:
It seems you didn't configure properly the toolchain, he's not able to find it.

shillpa-j, 2019-10-17:
Okay. Thanks for the response.
Can i know how to configure it?

haugoug, 2019-10-17:
From the README:
## Dependencies

### Build

Have a look at the dependencies documentation to see how to build them.

You can have a look [here](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain.git) for the toolchain.

### Setup

All the dependencies required to build the SDK must be setup through environment variables.

The toolchain must be built separately and the following environment variable should 
point to it:

    $ export PULP_RISCV_GCC_TOOLCHAIN=<path to the folder containing the bin folder of the toolchain>

RTL platforms should also be built separately (see the platform documentation for that) and the following
environment variable must point to the folder where the platform was installed (this example is for pulpissimo):

    $ export VSIM_PATH=<pulpissimo root folder>/sim

shillpa-j, 2019-10-17:
HI sir..
For the "RUNTIME BUILD" Installation we have given the same path as that of
we gave for the GVSoC build which was its prior step. And I even tried the
setup that you have sent but still i ended up getting the same error.

Thanks and Regards
Shilpa

haugoug, 2019-10-23:
Please configure PULP_RISCV_GCC_TOOLCHAIN

closed 2019-10-23 by haugoug