pulp-platform/pulpissimo #95: "make clean build " fails with Modelsim

shillpa-j, 2019-10-21:
Hi...
I have been trying to install the RTL simulation paltform for the PULPissimo. But, for the command "make clean lib build opts"  I'm facing the following error

error text:
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/pulpissimo_lib
Error: cannot find "/root/pulp-sdk/pulpissimo/sim/altera/16.0/modelsim_ase/bin/../linux_rh60/vlib"
vcompile/build.mk:56: recipe for target '/root/pulp-sdk/pulpissimo/sim/modelsim_libs/L2_tcdm_hybrid_interco_lib' failed
make[2]: *** [/root/pulp-sdk/pulpissimo/sim/modelsim_libs/L2_tcdm_hybrid_interco_lib] Error 1
/root/pulp-sdk/pulpissimo/sim//vcompile/ips.mk:53: recipe for target 'lib' failed
make[1]: *** [lib] Error 2
Makefile:37: recipe for target 'lib' failed
make: *** [lib] Error 2
 
Can I please know how to rectify this error?

Thanks and Regards 
Shilpa

bluewww, 2019-10-21:
Did you run `update-ips` before calling `make`?

shillpa-j, 2019-10-21:
Yes, I did

bluewww, 2019-10-21:
This is the error: `Error cannot find: "/root/pulp-sdk/pulpissimo/sim/altera/16.0/modelsim_ase/bin/../linux_rh60/vlib"`.

 It looks to me like your `modelsim` installation has some issues. Maybe bad path or something?

shillpa-j, 2019-10-22:
I dint understand that bad path. Should I reinstall the modelsim?

quangdaovu, 2019-10-22:
You better search Google for "Linux PATH environment variable". Your PATH is kinda bad.

shillpa-j, 2019-10-22:
Ok, I'll check it out. Thanks for the response

shillpa-j, 2019-10-22:
i have a query where exactly the modelsim has to be installed. I mean like
whether inside the sim folder of pulpissimo or seperately

quangdaovu, 2019-10-22:
You might want to install it seperately. ModelSim is a tool that is used for every project, not just pulpissimo.

shillpa-j, 2019-10-22:
Hi,

I have installed the Modelsim seperately and tried to add it in the path as
follows:

root@reva-HP-EliteBook-745-G2:~/pulp-sdk/pulpissimo# export
PATH=/root/altera/16.0/modelsim_ase/bin:$PATH
root@reva-HP-EliteBook-745-G2:~/pulp-sdk/pulpissimo# which vsim
/root/altera/16.0/modelsim_ase/bin/vsim

I hope this procedure is right.
After this i tried running the commands:

source setup/vsim.sh
cd sim/
make clean lib build opts

For this i got the following error:

root@reva-HP-EliteBook-745-G2:~/pulp-sdk/pulpissimo# source setup/vsim.sh
root@reva-HP-EliteBook-745-G2:~/pulp-sdk/pulpissimo# cd sim
root@reva-HP-EliteBook-745-G2:~/pulp-sdk/pulpissimo/sim# make clean lib
build opts
rm -rf
/root/pulp-sdk/pulpissimo/sim/modelsim_libs/L2_tcdm_hybrid_interco_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/adv_dbg_if_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/apb2per_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/apb_adv_timer_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/apb_fll_if_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/apb_gpio_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/apb_node_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/apb_interrupt_cntrl_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/axi_node_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/axi_slice_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/axi_slice_dc_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/axi_mem_if_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/timer_unit_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/common_cells_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/fpu_div_sqrt_mvp_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/fpnew_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/jtag_pulp_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/riscv_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/ibex_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/scm_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/generic_FLL_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/tech_cells_generic_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_core_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_uart_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_i2c_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_i2s_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_qspi_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_sdio_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_camera_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_filter_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/udma_external_per_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/hwpe_ctrl_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/hwpe_stream_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/hwpe_mac_engine_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/riscv_dbg_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/pulp_soc_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/tbtools_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/tb_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/vip_lib
rm -rf /root/pulp-sdk/pulpissimo/sim/modelsim_libs/pulpissimo_lib
Error: cannot find "/root/altera/16.0/modelsim_ase/bin/../linux_rh60/vlib"
vcompile/build.mk:56: recipe for target
'/root/pulp-sdk/pulpissimo/sim/modelsim_libs/L2_tcdm_hybrid_interco_lib'
failed
make[2]: ***
[/root/pulp-sdk/pulpissimo/sim/modelsim_libs/L2_tcdm_hybrid_interco_lib]
Error 1
/root/pulp-sdk/pulpissimo/sim//vcompile/ips.mk:53: recipe for target 'lib'
failed
make[1]: *** [lib] Error 2
Makefile:37: recipe for target 'lib' failed
make: *** [lib] Error 2

Can I know what is wrong in this?

closed 2019-10-22 by shillpa-j

bluewww, 2019-10-22:
Could you fix this issue?

shillpa-j, 2019-10-22:
Sorry i dint get that

quangdaovu, 2019-10-22:
So your ModelSim installation directory is /root/altera/16.0/modelsim_ase/. Now open /root/altera/16.0/modelsim_ase/vco with text edittor, find 

```
*)                vco="linux_rh60" ;;
```

and change to

```
*)                vco="linux" ;;
```
Save the file and try again.

shillpa-j, 2019-10-23:
yes. I tried and this issue has been solved but the problem was i'm getting another error for the same command and I have posted as another issue in the community .
  
Thanks and Regards
shilpa