pulp-platform/pulpissimo #96: Modelsim issue

shillpa-j, 2019-10-22:
Hi,
Currently i have been working on PULPissimo installation and I have installed Modelsim 10.4d, since 10.6c wasn't available. Is there any probelm of using 10.4d version instead of 10.6c as the requirements mention that modelsim 10.6c is recommended.

I'm getting an error as follows for the modelsim 10.6c

Modifying /root/altera/16.0/modelsim_ase/linuxaloem/../modelsim.ini
Building L2_tcdm_hybrid_interco/soc_interconnect
Building adv_dbg_if/adv_dbg_if
Building apb2per/apb2per
Building apb_adv_timer/apb_adv_timer
Building apb_fll_if/apb_fll_if
Building apb_gpio/apb_gpio
Building apb_node/apb_node
Building apb_interrupt_cntrl/apb_interrupt_cntrl
Building axi_node/axi_node
Building axi_slice/axi_slice
Building axi_slice_dc/axi_slice_dc
Building axi_mem_if/axi_mem_if
Building timer_unit/timer_unit
Building common_cells/common_cells_all
** Warning: ../ips/common_cells/src/fifo_v3.sv(144): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/common_cells/src/fifo_v3.sv(148): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: ../ips/common_cells/src/fifo_v3.sv(13): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Error: ../ips/common_cells/src/id_queue.sv(268): Questa has encountered an unexpected internal error: ../../src/vlog/vgentd.c(515). Please contact Questa support at http://supportnet.mentor.com/
/root/pulp-sdk/pulpissimo/sim/vcompile//ips/common_cells.mk:73: recipe for target '/root/pulp-sdk/pulpissimo/sim/modelsim_libs/common_cells_lib/common_cells_all.vmake' failed
make[2]: *** [/root/pulp-sdk/pulpissimo/sim/modelsim_libs/common_cells_lib/common_cells_all.vmake] Error 2
/root/pulp-sdk/pulpissimo/sim//vcompile/ips.mk:14: recipe for target 'build' failed
make[1]: *** [build] Error 2
Makefile:33: recipe for target 'build' failed
make: *** [build] Error 2

Thanks and Regards
Shilpa

bluewww, 2019-10-22:
Looks like modelsim dies in the `common_cells` library. This is due to your modelsim version being old. I recommend you update modelsim to a newer version. We mostly run PULPissimo with new versions of modelsim and rely on community patches for older version.

FrancescoConti, 2019-10-22:
The assert that most likely is causing this issue (https://github.com/pulp-platform/common_cells/blob/b86af02f630977efe570961444583cac03d8e2c2/src/fifo_v3.sv#L143 and following lines) is checking that the FIFO is consistent and no write is performed when it’s full, no read when it’s empty. you can safely  comment it out, although you will get a dangerous silent error (like you would in real HW). You can also ``define VERILATOR` to remove other similar blocks all together.

shillpa-j, 2019-10-23:
Hi...
I have installed altera modelsim 10.4d version but I'm unable to get the
10.6c version. Would anyone please provide Modelsim 10.6c link for ubuntu
16.04 as I'm unable to get it on Internet.

Regards
Shilpa

bluewww, 2019-10-23:
Sorry we cant help you with this. You have to contact altera for a newer version of yoir simulator.

closed 2019-10-23 by bluewww