pulp-platform/pulpissimo #102: Ibex: got stuck in `rt_event_yield` function

midlll, 2019-11-15:
Hi all,
I tried running an example on Ibex:
```
#include <stdio.h>
#include <rt/rt_api.h>
#include <stdint.h>

int __rt_fpga_fc_frequency = 20000000; // e.g. 20000000 for 20MHz;
int __rt_fpga_periph_frequency = 10000000; // e.g. 10000000 for 10MHz;

int main()
{
  //while(1){
    printf("Hello! \n\r");
    printf("World! \n\r");
    //for(int i=0; i<1000000000; i++);
  //}
  return 0;
}
```
It got stuck after first `printf` function.
I tried debug it and it goes in a loop in `rt_event_yield()` function.
```
__rt_io_uart_flush (debug_struct=0x1c000e04 <__hal_debug_struct>) at libs/io/io.c:343
343	      while(__rt_io_pending_flush)
(gdb) 
345	        rt_event_yield(NULL);
(gdb) 

^C
Program received signal SIGINT, Interrupt.
__rt_no_irq_handler () at kernel/riscv/rt/crt0.S:220
220	  j __rt_no_irq_handler
(gdb) 
```

I also tried [timer_oneshot](https://github.com/pulp-platform/pulp-rt-examples/blob/master/time/timer_oneshot/test.c) example from `pulp-rt-examples`, and it got stuck in the same function:
```
main () at test.c:38
38	  rt_timer_start(&timer, 1000000);
(gdb) 
41	  while(nb_ticks < NB_TICKS)
(gdb) 
38	  rt_timer_start(&timer, 1000000);
(gdb) 
41	  while(nb_ticks < NB_TICKS)
(gdb) 
43	    rt_event_yield(NULL);
(gdb) 

^C
Program received signal SIGINT, Interrupt.
__rt_event_execute (sched=<optimized out>, wait=wait@entry=1) at kernel/events.c:200
200	      rt_irq_enable();
(gdb) 
```
I'm not really sure what `rt_event_yield()` function do, just that it waits for events to be executed.

### Specs

I'm using Xilinx Zedboard with Zynq-7000; FPGA bitstream with Ibex configuration.
Steps:
```
cd pulp-platform/pulp-sdk/
make clean
source configs/pulpissimo_ibex.sh 
source configs/fpgas/pulpissimo_ibex/genesys2.sh 
make all
source pkg/sdk/dev/sourceme.sh
cd ../pulp-rt-examples/time/timer_oneshot/
make clean all
$PULP_RISCV_GCC_TOOLCHAIN/bin/riscv32-unknown-elf-gdb build/pulpissimo/test/test
```

I tried running all this on Ri5cy core and it ran smoothly.

Do you know maybe what the problem is here?

Thanks and kind regards.

vogelpi, 2019-11-18:
Hi @midlll ,
I am not 100% sure but I think this could be an issue with the interrupt IDs.

Ibex uses a RISC-V standard-compliant interrupt framework. It supports one timer (mtip), software (msip), external (meip), as well as 15 fast interrupts (mfip 0 - 14).

In PULPissimo, the interrupts are connected to the fast interrupts (mfip 0 - 14). Internally to the core, the IDs of these interrupts are 16 - 30. Im not sure if the SDK uses the correct mapping. But the error
```
Program received signal SIGINT, Interrupt.
__rt_no_irq_handler () at kernel/riscv/rt/crt0.S:220
220	  j __rt_no_irq_handler
```
suggests that there is no handler installed for the interrupt received.

As quick and dirty hack (and also to check if this is really the problem), you can try to change the internal ID inside Ibex. Inside `ibex_controller.sv` you would need to change Line 407 from
```exc_cause_o = exc_cause_e'({2'b11, mfip_id});```
to
```exc_cause_o = exc_cause_e'({2'b10, mfip_id});```
Please let us know if this works. We can then try to change the SDK.

Best,
Pirmin

MarekPikula, 2019-11-18:
I stumbled upon the same issue. I'm working on a patch to fix it without changing the SDK.

midlll, 2019-11-18:
Hi @vogelpi ,
I changed the ID and now it works for the first "Hello!" example :).  

However in below example `timer_oneshot` is the same. So I tried some more examples like `time/wait_time` and `gpio/output` and it all stops in the same place:
```
Program received signal SIGINT, Interrupt.
__rt_event_execute (sched=<optimized out>, wait=wait@entry=1) at kernel/events.c:200
200	      rt_irq_enable();
```

Thanks,
Mirnesa

midlll, 2019-11-20:
I reopened this issue because the previous solution worked partially, juts to go through program without any `rt-function` that wait for some event. If I use for example `rt_time_wait_us()` function, it gets stuck again.  
This works:
```
while(1){
    printf("Hello! \n\r");
    rt_gpio_set_pin_value(0, GPIO, 1);
    for(int i=0; i<10000000; i++);
    //rt_time_wait_us(2000000);
    printf("World! \n\r");
    rt_gpio_set_pin_value(0, GPIO, 0);
    //rt_time_wait_us(2000000);
    for(int i=0; i<10000000; i++);
  }
```

But if I use `rt_time_wait_us()` it doesn't go through it:
```
while(1){
    printf("Hello! \n\r");
    rt_gpio_set_pin_value(0, GPIO, 1);
    //for(int i=0; i<10000000; i++);
    rt_time_wait_us(2000000);
    printf("World! \n\r");
    rt_gpio_set_pin_value(0, GPIO, 0);
    rt_time_wait_us(2000000);
    //for(int i=0; i<10000000; i++);
  }
```
I think it waits for some event that never occurs.

I'm a newbie in low level programming, so I'm not really sure what's exactly happening in the background.

haugoug, 2019-11-20:
The porting of the runtime to pulpissimo ibex is actually still on-going, which is why you get such issues. We hope to finish it soon, we'll let you know.

MarekPikula, 2019-11-20:
There is an interesting issue open for ibex: lowRISC/ibex#482

Maybe it can help you understand how interrupts work when multiple are active.

bluewww, 2021-02-02:
This is outdated in light of the upcoming ibex support/changes

closed 2021-02-02 by bluewww