pulp-platform/pulpissimo #104: HWPE Example Hangs on ZedBoard

byteoctopus, 2019-11-28:
I generated a bitstream for the ZedBoard with USE_HWPE=1 in xilinx_pulpissimo.v and am now trying to run the HWME example from pulp-rt-examples.

The demo hangs on the call to __rt_periph_wait_event(ARCHI_SOC_EVENT_FCHWPE0, 1). How can I best go about debugging this?

bluewww, 2019-11-28:
I personally run it in a simulator and try to understand why the event `ARCHI_SOC_EVENT_FCHWPE0` was not raised.

haugoug, 2019-11-28:
this was changed recently and set to 140, make sure you have a recent runtime

TolleWurst, 2019-12-09:
> this was changed recently and set to 140, make sure you have a recent runtime

This used to work fine like 2 or 3 weeks ago but for some reason it doesn't work anymore.
I am using PULPissimo and simulate with QuestaSim.
For some reason the HWME example won't terminate.
It does the calculation and after that evt_o is 1 but still it doesn't terminate.

bluewww, 2021-02-02:
Should be fixed

closed 2021-02-02 by bluewww

Fatalon, 2021-05-27:
i still have the same issue - working on a Zedboard and using the current master of Pulpissimo. Could it be that the applied changes to the sdk are currently not included in the pulpissimo setup? Just from checking the version inside ips_list.yml - which is tied to v2.0.1 - changes for example made to update the hwpe versions https://github.com/pulp-platform/pulp_soc/pull/49/commits are just released in version v2.1.0. Is it safe to use this newer version of the sdk and does it contain the fix?