pulp-platform/pulpissimo #105: Where to find Pulpissimo tests?

aelgogary, 2019-11-30:
Hi, when i run update-tests script, it asks for a password for Gitlab, is there a copy of the repositories pulp-tests and pulp-sw on Github?

bluewww, 2019-11-30:
@haugoug

haugoug, 2019-12-01:
This an internal script that we use for non-regression runs, the repositories are not accessible.

Toxic-Scofield, 2019-12-09:
Hi, "This an internal script that we use for non-regression runs, the repositories are not accessible."  Does that mean we can not use these tests?

haugoug, 2019-12-09:
Yes we haven't had time yet to open-source these tests, they cannot be used.

bluewww, 2020-11-17:
We recently added some tests here
https://github.com/pulp-platform/regression_tests
but we don't refer to them yet in the ./update-tests-scripts

closed 2020-11-17 by bluewww