pulp-platform/pulpissimo #106: i2s simulation fails with DPI error

vikramjain236, 2019-12-02:
I tried to run the i2s code on the RTL simulation platform by including the i2s model from the vendor and using this to generate the scripts:
./generate-scripts --i2s-vip --i2c-vip --rt-dpi
However, when I run the i2s test from the pulp-rt-example I get the following error:

# 10.7c

# source /volume1/users/vjain/pulpissimo/sim/tcl_files/config/run_and_exit.tcl
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do
# couldn't read file "/volume1/users/vjain/pulpissimo/sim/tcl_files/disable_tcheck_fll.do": no such file or directory
# source /volume1/users/vjain/pulpissimo/sim/tcl_files/run.tcl
# ** Error: (vsim-3763) ****SystemVerilog DPI cannot access file '/volume1/users/vjain/pulp-builder/install/ws/lib/libpulpdpi.so'****
# No such file or directory. (errno = ENOENT)
# Use the -help option for complete vsim usage.
# vsim -c -quiet vopt_tb -L models_lib -L vip_lib -t ps "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+TB_PATH=/volume1/users/vjain/pulpissimo/sim" "+UVM_NO_RELNOTES" -permit_unmatched_virtual_intf -sv_lib /volume1/users/vjain/pulp-builder/install/ws/lib/libpulpdpi "+VSIM_PATH=/volume1/users/vjain/pulpissimo/sim" -gUSE_SDVT_SPI=0 -gUSE_SDVT_CPI=0 -gBAUDRATE=115200 -gENABLE_DEV_DPI=0 -gCONFIG_FILE=rtl_config.json -gLOAD_L2=JTAG -gUSE_SDVT_I2S=0 
# Error loading design
Error loading design
# Errors: 1, Warnings: 0
Simulator reported an error, leaving
make: *** [run] Error 1

haugoug, 2019-12-02:
This example is not using the DPI model that you get with option --i2s-vip , it is using the one you get with option --rt-dpi.
You can use the one your prefer, but we don't have any example for --i2s-vip.
If you want to use the other one, first remove the option --i2s-vip and recompile the platform. Then install the DPI models using pulp-builder with these commands:
./scripts/update-gvsoc
./scripts/build-gvsoc
Then you can relaunch.

vikramjain236, 2019-12-02:
Hi @haugoug , I couldn't get it working even after doing what you said. This is the new error I get:

 make clean all run
rm -rf /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv//bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c test.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv//bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv//bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -Wextra -Wall -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wundef -fdata-sections -ffunction-sections -I/volume1/users/vjain/pulp-builder/install/include/io -I/volume1/users/vjain/pulp-builder/install/include -include /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.c -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test`
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv//bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/fc//volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o -nostartfiles -nostdlib -Wl,--gc-sections -L/volume1/users/vjain/pulp-builder/install/rules -Tpulpissimo/link.ld -L/volume1/users/vjain/pulp-builder/install/lib/pulpissimo -lrt -lrtio -lrt -lgcc
**/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/../lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lrt
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/../lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lrtio
/volume1/users/vjain/pulp-riscv-gnu-toolchain/output/riscv/bin/../lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lrt
collect2: error: ld returned 1 exit status
make: *** [/volume1/users/vjain/pulp-rt-examples/hello/build/pulpissimo/test/test] Error 1**

haugoug, 2019-12-09:
It seems the runtime has not been built, can you try this command:
scripts/build-runtime

vikramjain236, 2019-12-12:
Yes, there was a problem with the build which I fixed. But, I get a new set of errors:
# This port will contribute value (U) to the signal network.
# ** Warning: (vsim-3040) Command line generic/parameter "USE_SDVT_I2S" not found in design.
# ** Warning: (vsim-3040) Command line generic/parameter "USE_SDVT_SPI" not found in design.
#  run_and_exit
# [TB]                    0 - Found TB driver component (index:           0, name: uart, type: dpi)
# [TB]                    0 - Instantiating DPI component
# [TB]                    0 - Starting task id           0
# [TB]                    0 - Starting task id           1
# [TB]                    0 - Got interface information (index:           0, name: uart, type: UART, id:           0, sub_id:           0)
# [TB]                    0 - Found TB driver component (index:           1, name: jtag_proxy, type: dpi)
# [TB]                    0 - Instantiating DPI component
# ** Warning: (vsim-PLI-8496) $fatal : Argument number 1 is invalid. Expecting 0, 1, or 2. Using default value of 1
#    Time: 0 ps  Iteration: 0  Process: /tb_pulp/genblk1/#INITIAL#272(#ublk#176585792#273) File: /volume1/users/vjain/pulpissimo/sim/../rtl/tb/dpi_models/dpi_models.sv Line: 227
# ** Fatal: [TB]                    0 - ERROR, Failed to open periph model (jtag_proxy.so) with error: /esat/micas-data/software/Mentor/questasim_10.7c/questasim/gcc-5.3.0-linux_x86_64/lib64/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by /volume1/users/vjain/pulp-builder/install/ws/lib/jtag_proxy.so)
#    Time: 0 ps  Scope: dpi_models.dpi_fatal File: /volume1/users/vjain/pulpissimo/sim/../rtl/tb/dpi_models/dpi_models.sv Line: 227
# ** Note: $finish    : /volume1/users/vjain/pulpissimo/sim/../rtl/tb/dpi_models/dpi_models.sv(227)
#    Time: 0 ps  Iteration: 0  Region: /tb_pulp/genblk1
# End time: 11:49:12 on Dec 12,2019, Elapsed time: 0:00:01
# Errors: 5, Warnings: 37

closed 2019-12-17 by haugoug