pulp-platform/pulpissimo #108: Error：OpenOCD only supports Debug Module version 2, not 1 (dmstatus=0x1)

Toxic-Scofield, 2019-12-04:
Hi，
    I want to implement pulpissimo on ZC706. Now I have been ready to emulate PULPissimo on ZC706. However, when I launch openocd with configuration file as follow as an argument，I met some errors.
[image]

### My configuration file is as follow:

    //JTAG-smt2 (from upstream OpenOCD)

    interface ftdi
    ftdi_vid_pid 0x0403 0x6014

    ftdi_layout_init 0x2088 0x3f8b
    ftdi_layout_signal nSRST -data 0x2000
    ftdi_layout_signal GPIO2 -data 0x2000
    ftdi_layout_signal GPIO1 -data 0x0200
    ftdi_layout_signal GPIO0 -data 0x0100
    ftdi_device_desc "Digilent USB Device" 

    //PULPissimo

    adapter_khz     1000

    set _CHIPNAME riscv

    jtag newtap $_CHIPNAME unknown0 -irlen 5 -expected-id 0x10102001
    jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511C3

    set _TARGETNAME $_CHIPNAME.cpu
    target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0x3e0

    gdb_report_data_abort enable
    gdb_report_register_access_error enable

    riscv set_reset_timeout_sec 120
    riscv set_command_timeout_sec 120

    //prefer to use sba for system bus access
    riscv set_prefer_sba on

    //dump jtag chain
    scan_chain

    init
    halt
    echo "Ready for Remote Connections"

### I have install necessary dependencies.
Do you know maybe what the problem is here?
Thanks and best regards!

Toxic-Scofield, 2019-12-09:
R E S E T！ O K ! ! !

closed 2019-12-09 by Toxic-Scofield

bluewww, 2019-12-09:
Could you resolve your issue? I'm about to give the OpenOCD stuff a look I know there are a few known problems.

Toxic-Scofield, 2019-12-11:
> Could you resolve your issue? I'm about to give the OpenOCD stuff a look I know there are a few known problems.

I've solved this. I didn't reset pulpissimo.

gsriker77, 2022-02-19:
Dear Toxic-Scofield,
Can you please provide solution what you did to resolve it?
Thanks
Sriker.G