pulp-platform/pulpissimo #111: can not open modelsim

ziang87, 2019-12-23:
Hello,everyone.I am a student from China.I followed the steps,and I meet some problems.
"
After having access to the SDK, you can build the simulation platform by doing the following:
source setup/vsim.sh
- cd sim/
- make clean lib build op
well,when I do this step,some problems occured.
root@cp1-VirtualBox:~# source setup/vsim.sh
bash: setup/vsim.sh: No such file or directory
 I did some research on the Internet, and it was said that it was because ubuntu was 64-bit, but I installed a 32-bit library file and still got an error.Can you help me?

bluewww, 2019-12-23:
You are probably in the wrong folder when you execute the `source` command. It says it right there:`bash: setup/vsim.sh: No such file or directory`. Execute `source setup/vsim.sh` in the root folder of the project.

bluewww, 2020-01-27:
The scripts were fixed to be independant of the location where you source them from. Please reopen if you still have this issue.

closed 2020-01-27 by bluewww