pulp-platform/pulpissimo #112: Ibex on FPGA

quangdaovu, 2019-12-26:
Hi,

I have encountered an issue with Ibex on FPGA recently. I built the default design of PULPissimo with CORE_TYPE set to 1, USE_FPU set to 0 on Genesys2 board. OpenOCD can recognize the core, I can load the application (hello in this situation) but the program failed to execute on ibex. 

In more details:
1. Testbench works fine. At first there are some 'illigal instruction', but after I generate new boot_code.cde then they're gone.
2. There is no timing violation on FPGA design.
3. OpenOCD recognizes the core and I can load the application using gdb.
4. The program failed to execute. There is 2 cases:
  - If I execute the program step by step (`load` then `s` on gdb) then the chip initialization process stucks at kernel/time.c line 156. It hangs at printf function, gdb shows 
```
unable to halt hart 992
  dmcontrol=0x83e00001
  dmstatus =0x00030c82
Remote failure reply: E0E
Remote failure reply: E0E
Remote failure reply: E0E
```
  - If I use `c` to execute the program after `load`, the core hangs at `__rt_illigal_instr`. Funny thing is breakpoint failed to work if I set the breakpoint to anywhere inside `__rt_init()` function. Meanwhile in the first case, `__rt_init()` is executed successfully.
5. I tried `pulp_builder` with both commit id `0e51ae60d66f4ec326582d63a9fcd40ed2a70e15` (as in README) and id `83953d5ca4c545f4186bf3683d509566c3067012` (as in Makefile). Config file is `configs/pulpissimo_ibex.sh`. I also modified the config file for FPGA on `configs/fpga/pulpissimo/genesys2.sh` from 
```
export PULP_CURRENT_CONFIG=pulpissimo@config_file=fpgas/pulpissimo/genesys2.json
```
to 
```
export PULP_CURRENT_CONFIG=pulpissimo_ibex@config_file=fpgas/pulpissimo_ibex/genesys2.json
```

Has anyone successfully to evaluate ibex on FPGA yet? Did I do something wrong?

Thank you in advance.
Regards.

quangdaovu, 2019-12-30:
Hi,

Here is some update:
- I can manage to make it works by modify `ibex_controller.sv` according to https://github.com/pulp-platform/pulpissimo/issues/102.
- Interrupt mapping seems still off. I have to change `ARCHI_FC_EVT_GPIO` from `15` to `19` for GPIO interrupt to work.

bluewww, 2021-02-02:
This is outdated in light of the upcoming ibex changes. Please re-open if this issue persists.

closed 2021-02-02 by bluewww