pulp-platform/pulpissimo #113: JTAG mode vectors in pulpissimo

lanceburton, 2019-12-31:
Hi,

in tb_pulp.sv, I can see code below. If I want run in JTAG mode, How can I genarate the vector file or where can I find this vector files?

Thanks!

              // read in the stimuli vectors  == address_value
               if ($value$plusargs("stimuli=%s", stimuli_file)) begin
                  $display("Loading custom stimuli from %s", stimuli_file);
                  $readmemh(stimuli_file, stimuli);
               end else begin
                  $display("Loading default stimuli");
                  $readmemh("./vectors/stim.txt", stimuli);
               end

bluewww, 2020-01-27:
Generally the stimuli file (vectors) are generated by the [pulp-sdk](https://github.com/pulp-platform/pulp-sdk9). If you want a more minimal setup you can use the [pulp-runtime](https://github.com/pulp-platform/pulp-runtime/). In the last repository you can also find [here](https://github.com/pulp-platform/pulp-runtime/blob/master/bin/stim_utils.py) a script to manually generate a `stim` file from your elf file.

lanceburton, 2020-03-12:
Thank you very much for your reply！

closed 2020-03-12 by lanceburton