pulp-platform/pulpissimo #114: Loading Flash via JTAG

TolleWurst, 2020-01-15:
Hello,

I am using PULPissimo and currently trying to load the S25FS256S flash via JTAG.
The goal is to be able to load an image, for example an image of the _hello_world_ test, into flash and execute it.
At the moment the pre-loading feature of the flash is used to already have the stimulus in the flash. If the standalone boot is used, the test is correctly executed and the expected "hello" is printed.
This should be changed so that instead of using the pre-loading feature, the stimulus is loaded into flash via JTAG.
I have tried to find information on how you would go about doing that and i'm unsure if this is possible at the moment (without changes or expansions to the jtag_pkg and maybe further changes).

What i found are a few issues regarding similiar topics:
https://github.com/pulp-platform/pulpissimo/issues/88
https://github.com/pulp-platform/pulpissimo/issues/36
https://github.com/pulp-platform/pulpissimo/issues/26

So now i am wondering, is it possible to load an image (for example of the _hello_world_ test, or any other image actually) into the S25FS256S flash via JTAG.
If yes, what would be the necessary steps to do so?
If no, what changes or extensions would be needed to be able to do so?

bluewww, 2020-01-17:
The flash memory `s25fs256s` is connected to the qspi. We currently don't have a script to flash the chip since we normally run the simulation with the chip already preflashed as you mentioned.

So the way to do this is to write data from the jtag to the L2 and then engage the udma to transfer the data to the flash. The debug module can access the whole address space, thus also the memory mapped udma.
@haugoug

LPLATU, 2020-03-04:
So after loading the relevant data into L2 using JTAG (either using _debug_mode_if.load_L2_ or _debug_mode_if.writeMem_) how would you recommend loading that data into the flash?

I'm thinking one way would be to write a program using the SPIM driver from the PULP Kernel Library. But this would require the core to do the work.

You said that the udma is memory mapped, so is there currently a way to to load the data from L2 to the flash while halting the core (maybe via SPI?), similar to using JTAG to load the data into L2 first?

Thank you,
LPLATU