pulp-platform/pulpissimo #118: Running Pulp-rt-examples on Zedboard

aelgogary, 2020-02-10:
Hi iam trying to run hello example on Zedboard with Digilent JTAG HS2, i followed the steps in the instruction but for some reason when loading the elf file using gdb it hangs and nothing is returned, her is my code and logs.
`
#include "rt/rt_data.h"
#include <stdio.h>
#include <rt/rt_api.h>
#include <stdint.h>

#define GPIO 5 // LED0 on Genesys2 board
#define LED4 4 // LED1 on Genesys2 board
unsigned int __rt_iodev_uart_baudrate = 115200;
int __rt_fpga_fc_frequency = 20000000;
int __rt_fpga_periph_frequency = 10000000;

int main()
{
 // GPIO initialization
 rt_pad_set_function(GPIO, 1); //pad_func=1 means gpio functionality. pad_func=0 is default (e.g. in this case spim_csn1)
 rt_gpio_init(0, GPIO);

 // Configure GPIO as an outpout
 rt_gpio_set_dir(0, 1<<GPIO, RT_GPIO_IS_OUT);

 // Now set the output value

 rt_gpio_set_pin_value(0, GPIO, 1);	

 // GPIO initialization
 rt_pad_set_function(LED4, 1); //pad_func=1 means gpio functionality. pad_func=0 is default (e.g. in this case spim_csn1)
 rt_gpio_init(0, GPIO);

 // Configure GPIO as an outpout
 rt_gpio_set_dir(0, 1<<LED4, RT_GPIO_IS_OUT);

 // Now set the output value

 rt_gpio_set_pin_value(0, LED4, 1);	

int i = 1000;
while(i>= 0){
printf("testing"); 
i = i - 1;
}
 rt_gpio_set_pin_value(0, LED4, 0);
 return 0;
}`

`$OPENOCD/bin/openocd -f openocd-zedboard-hs2.cfg 
Open On-Chip Debugger 0.10.0+dev-00615-g73c03b4 (2020-01-24-09:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 1000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
   TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 riscv.unknown0         Y     0x00000000 0x10102001     5 0x01  0x03
 1 riscv.cpu              Y     0x00000000 0x249511c3     5 0x01  0x03
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.unknown0 tap/device found: 0x10102001 (mfg: 0x000 (<invalid>), part: 0x0102, ver: 0x1)
Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1024 harts
Info :  hart 0: currently disabled
Info :  hart 1: currently disabled
Info :  hart 2: currently disabled
Info :  hart 3: currently disabled
Info :  hart 4: currently disabled
Info :  hart 5: currently disabled
Info :  hart 6: currently disabled
[... 1068 lines cut ...]
(gdb) 
do_ctors () at kernel/init.c:39
39	  for(fpp = ctor_list+1;  *fpp != 0;  ++fpp) {
(gdb) 

__rt_init () at kernel/init.c:145
145	  do_ctors();
(gdb) 
do_ctors () at kernel/init.c:40
40	    (**fpp)();
(gdb) 
`
also using (gdb) disassemble gives <__rt_illegal_instr>
```
Dump of assembler code for function _start:
=> 0x1c008080 <+0>:	j	0x1c0080a0 <_stext>
   0x1c008084 <+4>:	j	0x1c0080ec <__rt_illegal_instr>
   0x1c008088 <+8>:	unimp
   0x1c00808a <+10>:	unimp
   0x1c00808c <+12>:	unimp
   0x1c00808e <+14>:	unimp
End of assembler dump.
```

Thanks in advance

meggiman, 2020-02-24:
Did you source the right config file? You should use configs/fpgas/pulpissimo/genesys2.sh. It doesn't matter that you don't use the Genesys2 board. The only board specific setting this file contains is the frequency which you already override in your source file (__rt_fpga_fc_frequency...). Make sure to recompile the sdk after you sourced the new configuration and to also recompile the application.

aelgogary, 2020-03-12:
Thank you, i was able to fix the problem by sourcing the right files and then compile the sdk as you said.
source configs/pulpissimo.sh
source configs/fpgas/pulpissimo/genesys2.sh

closed 2020-03-12 by aelgogary

mohamedElbouazzati, 2021-04-09:
hello @aelgogary did you displayed anything with screen or minicom on terminal? can you please tell me how did you the uart connection with zedboard 
Thanks you