pulp-platform/pulpissimo #119: Run C code on ZCU104

AnouarNechi, 2020-02-25:
I have to say I am confused on how to test my code on the ZCU104. The documentation is not really clear regarding the steps (especially for Newbies).
I want to run some benchmarks using pulpissimo on the ZCU104. But I did not get how to use the pulp-sdk while it supports only the Genesys2 Board.
If I want only to use the Pulpissimo repository (without SDK) what are the steps to run my C code on the ZCU104.
I will appreciate your help.

FrancescoConti, 2020-02-25:
Having never done it myself: have you tried using the Genesys2 configuration in the PULP SDK? It might work on the ZCU104, too.

meggiman, 2020-02-25:
Hi @AnouarNechi 
I admit installing and using the SDK is not as straight forward and well documented as we want it to be but we are working on it. However, in order to run your own C code on Pulpissimo you need to use the pulp-sdk unless you want to write your own runtime to setup the hardware (FLL configuration, GPIO initialization etc.). Contrary to your assumption the **pulp-sdk also supports all the other FPGAs**. I guess your assumption originates from the fact that there is no custom-tailored configuration file like configs/fpga/pulpissimo/genesys2.sh for the other boards. So far there is no genesys2 specific configuration in this file and you can also use it to compile the SDK and your C-code for any of the other FPGA boards.  As soon as we will find the time to do so we will add custom config scripts for the other boards as well.
Since you will most likely be using OpenOCD in order to debug and preload your application into PULPissimo I suggest you install the standalone version of pulp-sdk from https://github.com/pulp-platform/pulp-sdk instead of using the simplified pulp-builder flow as the standalone version also includes the patched OpenOCD version.
Follow the installation steps in the README.md (i.e. installing the dependencies and the pulp-riscv-gcc toolchain). Don't forget to point to pulp sdk to where you installed the gcc toolchain by setting the environment variable. E.g.:
```
export PULP_RISCV_GCC_TOOLCHAIN=<path to the folder containing the bin folder of the toolchain>
```

 Afterward, run the following commands from within the pulp-sdk root directory) :
```
source configs/fpgas/pulpisssimo/genesys2.sh
```
To choose pulpissimo as the target and configure the C runtime for FPGA usage. **Again, it doesn't matter that you are not using the genesys2 board. It should work regardless.** 

Then run 
```
make all
``` 
This will build the sdk for PULPissimo running on fpgas.

In order to build the patched OpenOCD version go through the steps described in the GDB and OpenOCD section of the PULPissimo README.md. 

After this is done, source the newly generate sourceme.sh file and you are ready to compile your own applications (again, see FPGA section in README.md of the Pulpissimo project)