pulp-platform/pulpissimo #121: I2S Example Output

FrischAd, 2020-03-09:
Hey guys,
I am using the I2S vip and connectet it with the PULPissimo SoC. Then I tested the I2S example from you: [Example](https://github.com/pulp-platform/pulp-rt-examples/tree/master/periph/i2s/pdm_capture)
Could you provide the output of this example so I can validate my results?
How did you generate the i2s_buffer_0.hex?

Thanks in advance.

bluewww, 2020-03-10:
@hagoug do you know about this `hex` file?

haugoug, 2020-03-10:
This example is not using the I2S VIP and it has no expected value. It is supposed to be run with a real PDM microphone and is just capturing samples and displaying them.
For what concerns the I2S VIP, it's better if you just have a look in the model, I guess this model is pretty simple.

FrischAd, 2020-03-11:
Thank you haugoug for the explanation.
I have two questions regarding the VIP and the I2S module:
1. As far as I have seen the I2S VIP is able to provide a PDM signal. So it should be possible to use the VIP instead of a real PDM microphone, shouldn't it?
2. Just to be sure: Is the I2S interface of PULPissimo able to support PCM with I2S? And this should also be possible with the VIP.

haugoug, 2020-03-11:
For the I2S interface, yes it supports both PCM and PDM, but for the VIP I have no idea, I've never used it.

closed 2020-03-11 by FrischAd