pulp-platform/pulpissimo #123: Work in an isolated server with only Xcelium installed

ninipa, 2020-03-27:
Hi,
This is a pure question.
As described in the issue title, I will be working on a server with no root permission and limited networking access, but on that server, Xcelium is installed.
I tried but it's too hard to build the whole platform including all the dependencies. Instead I've built up the platform on my personal laptop (ubuntu16.04). I have:
compiled toolchain,
pulpissimo platform (rtl, ips, sim, ips_list, rtl_list and so on)
testcases (pulp-rt-examples)
Then I was blocked at "Building the RTL simulation platform -> make build" since I don't have any simulator on my laptop. Therefore I can't go through the whole flow.
Fortunately, I see there is Xcelium support in ./sim. So I'm thinking about copy the whole platform from my laptop to the isolated server I mentioned. Then do the simulation with Xcelium.
I'm wondering it that a possible way? BTW, OS on that server is REDHAT7, i'm not sure whether there is compatibility issue on the compiled sw toolchain?(it's compiled in ubuntu1604). And I think at least in theory there is no dependency issue on RTL simulation, right?
Thank you and waiting for the answers!

bluewww, 2020-03-27:
Yes this is very much possible. I also develop on a remote centos7 machine using ssh. You have think a bit how you want to handle gui sessions by using either X11 forwarding or VNC.

I added a small patch to here https://github.com/pulp-platform/pulpissimo/pull/124 so you can just call `make build-incisive`. There is no sdk support (as of now) for using xcelium so you have to figure out what parameters to feed to the simulator.

ninipa, 2020-03-28:
Really appreciate your quick response and the patch for incisive! I will try it later soon.
BTW, is the patch only for incisive or it's also needed for Xcelium?
In addition, could you please clarify a little bit about "you have to figure out what parameters to feed to the simulator"? For example, if run helloworld case, what specific thing I need to do? (Sorry I'm quite fresh about the env so maybe these questions are stupid...)

bluewww, 2020-03-28:
You could have a look at the Makefile in `pulpissimo/sim/Makefile.incisive`. It has a target called `sim-sc` (simulate using one host cpu core) that already sets all but one paramater/plusarg correctly. The missing plusarg, called `+stimuli`, is there to point to a file that is a sdk generated dump of the compiled executable (normally called `stim.txt`).

So what you need to do is to compile an executable with the sdk or the minimal [pulp-runtime](https://github.com/pulp-platform/pulp-runtime/) (try the latter first, it is easier). The output will be a `stim.txt` file. Now you can invoke the Makefile with `+stimuli` set. This will look most likely like this
```bash
(in the sim dir)
make -f Makefile.incisive XRUN_FLAGS="+stimuli=path_to_your_stim.txt"
```

ninipa, 2020-03-30:
Thanks very much! That's pretty clear.
I tried compile the RTL by "make build-sc" but got an error says package dm could not by bound in pulp_soc.sv line 16.
I grep the expected package in the whole "pulpissimo" directory but didn't find it...
Am I missing anything?

bluewww, 2020-03-30:
This package should be in `riscv-dbg`

ninipa, 2020-04-02:
Hi,
I re-built the whole environment and the irun elaboration is fine now. Looks like I missed something when I download ips.
Then, a new problem is that I can't figure out how to get stim.txt...
What I did is:
1. download pulp-runtime/
2. source ./configs/pulpissimo.sh
3. download pulp-rt-examples/
4. in hello, do "make all"
Code compilation is done but I can't find "stim.txt" in generated "build"
Did I miss anything?
Can you please guide me how to generate stim.txt? Thank you!

bluewww, 2020-04-02:
In https://github.com/pulp-platform/pulp-runtime/blob/da4157a0e22f9564c29a7594baccd83b556d9297/rules/pulpos/default_rules.mk you can see how the `stim.txt` is generated. The important command is ```$(PULPRT_HOME)/bin/stim_utils.py --binary=$(TARGETS) --vectors=$(TARGET_BUILD_DIR)/vectors/stim.txt```.

I'll see to adding a Makefile target for it or something similar.

ninipa, 2020-04-03:
Thanks very much! Looks like I also need to do "make run" along with "make all". Although it fails at modelsim stuff (I don't have modelsim), stim.txt is generated successfully.
And I've successfully run "hello" case with xrun:) Let me play with the design more. 
My goal is to porting this environment on Emulator.

ninipa, 2020-04-03:
Hi,
One more question - is there any document which introduces the Soc in detail? For example, my first goal is to replace tb with my own SPI flash model (to load my code) and my uart terminal model, these are just peri outside pulpissimo SoC. Then how about the subsystems in Soc, at least the functionality of each submodels in pulp_soc? Glad to know more about it:)

bluewww, 2020-11-17:
The best we have in the pulpissimo datasheet. It was updated a few months ago and should be more complete.

closed 2020-11-17 by bluewww

tanbaobaolove, 2023-04-20:
> Yes this is very much possible. I also develop on a remote centos7 (which is really basically redhat7) machine using ssh. You have think a bit how you want to handle gui sessions by using either X11 forwarding or VNC.
> 
> I added a small patch to here #124 so you can just call `make build-incisive`. There is no sdk support (as of now) for using xcelium so you have to figure out what parameters to feed to the simulator.

Could you please tell me where the Makefile.incisive is, I can't find it right now.