pulp-platform/pulpissimo #125: Compilation issue with <math.h> function

asyms, 2020-03-28:
Hi,

I've installed the pulpissimo platform and gcc toolchain and been writing some C code to test if everything is working correctly. For this, I've started from the pulp-rt-examples (https://github.com/pulp-platform/pulp-rt-examples), starting from the "hello" code.

Using functions from the standard <math.h> library is giving me some weird compilation issues. As an example function I'll take the "exp(..."). Invocating this function constant values or non-array variables works fine. The issue starts when I do something like this:

```
#include <stdio.h>
#include <math.h>

void tstfnc(double in[2]) {
	printf("in[0] = %f\n", in[0]);
	printf("in[1] = %f\n", in[1]);
}

int main() {	
	double test1 = 1.0;
	double test2[2] = {2.0,3.0};
	double test3[2];

	printf("test1 = %f\n", test1);
	printf("test2[0] = %f\n", test2[0]);
	printf("test2[1] = %f\n", test2[1]);

	// Calling the exp function on a array value doesn't work
	double val = test2[0];
	test3[0] = exp(val);

	// Call unrelated tstfnc function on test2 yields the error
	tstfnc(test2);

	return 0;
}
```

The most important to note is that without the (seemingly) unrelated function call at the bottom is what causes the error, kind of? Commenting this function call compiles without a problem and I get the expected exponentiated values if I were to print them. Compiling this regularly with gcc, with the function call present, yields no errors.
The exact error I'm getting upon building for the pulpissimo platform:
"Undefined reference to 'exp', and it traces back to the function "tstfnc". 

At first I thought the math.h linking might have been wrong, but without the function call there isn't an issue to invoke the "exp(...)" function or any other of the math functions.

Any help would be greatly appreciated!
Kind regards.

bluewww, 2020-03-28:
I mean this definitely sounds like a linking bug. I can't really reproduce your problem.

asyms, 2020-03-28:
-Wouldn't that be odd considering the call works fine when the following function isn't called?

-So you can compile the above code just fine?

-Is there any special compiling options I should specify in my makefile for the math.h to be properly linked?

bluewww, 2020-03-29:
Seems to be an optimization issue. When you remove your function the optimizer realizes that `exp` has no side effect (result discarded) and removes the function altogether so you get no linking error. When you add your function gcc somehow believes that it still needs the result of `exp` during link time so you get the missing symbol error.

The gist of it is you need to link against `-lm` when using `<math.h>`. So the link command should has `-lrt -lrtio -lrt -lgcc` but it should look like `-lm -lrt -lrtio -lrt -lgcc`.

asyms, 2020-03-29:
Thanks for the reply! 
That indeed does make sense somehow. 
How would I properly add this -lm to the command, i.e. in what config/setup file should the option be added?

bluewww, 2020-03-29:
I ran the command manually but adding `PULP_LDFLAGS=-lm` to your Makefile also works.

asyms, 2020-03-29:
That indeed did the trick. Thanks alot!

One final question maybe, how is it possible that without a flag it could calculate a simple exp() of a constant value or even a non-array variable?

bluewww, 2020-03-29:
You never used the result of `exp` so there was nothing calculated.

closed 2020-03-29 by bluewww

asyms, 2020-03-29:
Correct, in the above code I didn't use the result.

But in other tests I for example printed `exp(2.0)`, which worked fine without linking the library and caused my initial confusion that -lm wasn't the issue.

bluewww, 2020-03-29:
For such an expression the compiler's constant folding pass will directly compute the result with compiler internal routines indeed. This is possible because `exp` is a standardized function (see C langauge standard) whose input output behavior is fully known.

asyms, 2020-03-29:
Thanks for the help!